joltwallet/esp_littlefs #2: Looking for opened file is O(N) with the current code (remove fd_used)

X-Ryl669, 2019-11-22:
Currently, as I understand your code, you're saving a bitmap of the file descriptor used (opened) and then each time we need to perform an operation on a path, you're iterating this bitmap to figure out if the file is actually opened (by comparing the given argument with all opened file name). Worst, this comparison is done with the filesystem locked.

Thus, every function with a `const char * path` argument is a O(N) string comparison.
Specifically, I'm talking about `esp_littlefs_get_fd_by_name` function usage.

I think this won't scale if N is going to larger than 20 or more.

I wonder why you are using this, instead of either a hash instead ?

# Solution 1
 Compute a hash of the given file name (via a CRC, or a simple string hash: `hash = (path[i] * prime_number + hash);`), and compare hashes (so at least the costly string comparison is avoided, unless the hash is matching).

That way, you can avoid the limit in the number of file descriptor used.

So basically, the code would look like:
```
static uint32_t compute_hash(const char * path) {
   uint32_t hash = 5381;
   char c;

   while (c = *str++)
        hash = ((hash << 5) + hash) + c; /* hash * 33 + c */
    return hash;
}

static int esp_littlefs_get_fd_by_name(esp_littlefs_t *efs, const char *path){
    uint32_t hash = compute_hash(path)
    sem_take(efs);
    for(uint8_t i=0; i < efs->max_files; i++){
        if (efs->files[i].hash == hash && strcmp(path, efs->files[i].path)) {
            ESP_LOGD(TAG, "Found \"%s\" at FD %d.", path, i);
            return i;
        }
    }
    sem_give(efs);
    ESP_LOGD(TAG, "Unable to get a find FD for \"%s\"", path);
    return -1;
}

```

Since the hash won't match for most files, you'll avoid the strcmp and you don't need to store a bitmap to check if the file descriptor is used. 
You'll need some house keeping here to set the `efs->files[].hash` to zero when it's closed or not used, but that's pretty simple to do in `close`.
When allocating file descriptors, you can iterate the `efs->files` array until you find a zero hash (will probably cost the same than checking the bitmap bit by bit like you're currently doing).

Also, if you intention is *not* to avoid dynamic memory allocation, then make the `efs->files` a linked list instead of an array so we can append to the list as many "descriptor" as we want (actually, have memory for). 
The file **descriptor** can't be the index in the list in that case (since the list will shrink upon deletion) but can be the pointer to the list's item. It's an opaque value anyway.

BrianPugh, 2019-11-23:
Your above interpretation on the code is correct.

I used the currently solution because its simple and I basically just mirrored the SPIFFS port. Also, for my usecases, I only have a few files open at a time. I also don't have to worry about hash collisions with this implementation. On the esp32, I imagine very few people have more than a few files open at a time; a linked list would use a lot more memory (8 bytes per entry vs 20 bits for the current static method).

Do you have a use-case where the current limitations are being hit? I imagine the performance gain would be very small, and would only impact deleting and renaming files.

BrianPugh, 2019-11-23:
Looking through the code, it doesn't seem obvious if I'm checking on file open (I should), I'll fix that and add more tests this weekend.

X-Ryl669, 2019-11-23:
In my use case, the filesystem is used for storing "pseudo" large files that can't be stored to NVS and that are obviously writable and don't change *too* often. This happens mainly during updates and error reporting. The read only files for the webserver in my code are already built in the flash (so I don't have to deal with versioning when updating the firmware).

Typically, I'm storing logs, error/crash dumps, user specific files. Currently, there's a task that's acting as a remote procedure call for a server and sends the files (or part of) whenever requested. There's no limit in the number of connections/queries here (that's probably wrong, but that's how it's made).
Thus, I can't ensure it'll be only X files open at a time, unless I'm closing each file after each request (can be done, but it implies opening it again for the next request and so on, I have to evaluate the overhead).

BTW, there's no hash collision in the scheme above since **only if** the hash matches the file path is compared (so it'll be true 99.999...% of the time and if it's false it's detected). If the code was C++, we could even use compile-time computed hash and ensure **zero** string comparison at all for compile-time known path.

Anyway, I'm a bit paranoid when I see O(M*N) algorithms with strings since the program becomes non deterministic anymore, one long (and unrelated) file name can change the execution time of whatever time critical task using the filesystem. If the file name is user-controllable, it's even worse.

BrianPugh, 2019-11-26:
I think you'll have issues regardless of filesystem if you implement it this way on the esp32. I would recommend queueing up file read requests to a single task and stream them from there. It'll also (probably) result in better performance.

Even though the code you submitted is fairly complete, I'd rather not additionally complicate the codebase for such a minor (unless you can show me it's not minor, you could add a benchmark to it in the unit-tests) performance boost.

closed 2019-11-28 by BrianPugh