joltwallet/esp_littlefs #7: Not thread-safe when CONFIG_LITTLEFS_USE_MTIME is enabled

htech-cs50, 2020-05-12:
We are seeing "corrupted dir pair" and "bad block" error messages when under conditions where multiple threads are accessing a LittleFS partition.  

For example:
`src/littlefs/lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}`
and:
`src/littlefs/lfs.c:2712:debug: Bad block at 0x63`

Traces lead these messages back to stat operations on files.  Further investigation led to noticing that in `vfs_littlefs_stat()`, the call to `vfs_littlefs_get_mtime()` isn't mutex protected, and adding sem_take/sem_give around that call cleared up a lot of those errors.  

To be safe, as a workaround, we have disabled `CONFIG_LITTLEFS_USE_MTIME` in order to avoid any other non thread-safe code related to the mtime feature, which we don't require at this time.

BrianPugh, 2020-05-12:
I'll investigate this, thanks for the report!

BrianPugh, 2020-05-13:
hopefully https://github.com/joltwallet/esp_littlefs/commit/e8444420e047b11c5b96c089650095f11a99cb01 solves your issue; I wasn't able to create a unit-test to test this. If you have any ideas on how to accomplish this in a reproducible manner, let me know. Thanks!

htech-cs50, 2020-05-13:
We don't currently have a unit-test to re-create the issue.  All of the unit tests included in this component pass regardless, even the one that tests tasks accessing the volume at the same time.  Possibly if the unit tests would have multiple tasks calling stat, since that is where the issue seems to lie.  As far as the fix goes, we will try it as time permits. After a quick visual inspection - I'm wondering about vfs_littlefs_utime() which also calls vfs_littlefs_get_mtime() and does seem to do so without any locks.  Might that also be an issue?

BrianPugh, 2020-05-17:
I just added calling `stat` from 2 different tasks on 2 different cores to the unit tests. Previous to my fix in e844442 I got a bunch of `Bad block at 0x??` and `Corrupted dir pair at {0x?, 0x?}` errors. With the fix all of those errors went away in the unit test. So now that there's a test, I'm going to mark this as resolved. If the issue comes back or you still have similar issues, please open another issue. Thanks!

closed 2020-05-17 by BrianPugh