joltwallet/esp_littlefs #10: option to disable logging failed to open errors

atanisoft, 2020-06-24:
is there a way to disable forced printing of log messages when opening a non-existent file other than silencing all of esp_littlefs?
example:
```
[CFG-FS] 8/900 kb space used
[CDI] Checking /cfg/cdi.xml...
E (2467) esp_littlefs: Failed to open file. Error  (-2)
[CDI] File /cfg/cdi.xml does not exist
[CDI] Updating /cfg/cdi.xml (len 6006)
[CDI] Registering CDI with stack...
E (2543) esp_littlefs: Failed to open file. Error  (-2)
Creating config file /cfg/config
[WiFi] Starting WiFi Manager task
```
The code that is being used has internal checks for file existence, size, etc and it would be nice to not have the extra log messages by default.

Perhaps having an option to move these to verbose logging instead or something in the KConfig to disable them?

BrianPugh, 2020-06-24:
Do you know how it's handled for esp-idf internal components? I could mimic that here.

In the meantime (less than ideal), you could:

```
#include "esp_log.h"

esp_log_level_set("esp_littlefs", ESP_LOG_NONE);
```

atanisoft, 2020-06-24:
The only problem with doing that is it will silence all log messages when some are more useful. For other VFS drivers I don't think they log at all when a file open fails or other errno response.

BrianPugh, 2020-06-24:
Looking at the SPIFFS port, you're right. I also notice that I'm not setting `errno` in these situations.

I'll lower the logging level of these logs to `verbose` and add some `errno` stuff this evening. Thanks!

atanisoft, 2020-06-24:
btw, thanks for creating this component. I'll be shifting over to using it soon on my projects :)

BrianPugh, 2020-06-25:
I began setting `errno` in commit 285e4de63cbd4ebf8b2582c96ab29fd2220a3a58. Luckily, LittleFS used POSIX error numbering (except negative).

I lowered the verbosity of level of errors that may happen under normal operation in d058ee73aba629b0b5500bbc8b56ba90bbf72234. I still kept "horribly wrong" things like fundamental programming errors, OOM errors, and other things like that at the `error` verbosity level. If you see any unwanted logs that occur during normal operation, let me know.

I'm going to go ahead and close this issue for now. If you discover more logs that should be lowered in verbosity level, we can reopen this issue.

closed 2020-06-25 by BrianPugh

atanisoft, 2020-06-25:
Thanks, errno is typically negative iirc so that is fine. I think any error condition that is unexpected should be logged for sure. If I find anything I'll log an issue for it. Thanks!