joltwallet/esp_littlefs #11: basePath="/"

lorol, 2020-06-26:
Hi Brian,

Can you please verify, I believe, for SPIFFS is OK to load it to basePath="/" but for LittleFS I had to use basePath="" instead.
The "/" compiles and runs, but webpages of my app go to "not-found". It could be related to the application itself but i think, the root "/" is somehow differently interpreted in lfs, and care should be taken anyway.
I would force the choice basePath="/" to be   basePath="" to solve my sketches, but is this a good way?

BrianPugh, 2020-06-26:
I can try and investigate this either this evening or this weekend and I'll add a unit-test for it. It sounds like undesirable behavior

lorol, 2020-06-26:
Thanks ...
See this, maybe related:
https://github.com/ARMmbed/littlefs/issues/138

BrianPugh, 2020-06-28:
Investigating this on branch `root_mount`. I've so far added a unit-test that confirms your findings.

The primary issue seems to be [this line in the esp-idf vfs component](https://github.com/espressif/esp-idf/blob/625bd5eb1806809ff3cc010ee20d1f750aa778a1/components/vfs/vfs.c#L78). I'll do a quick investigation on how/why spiffs handles it fine. It probably involves explicitly checking for `"/"` and then passing along a 0-length string.

BrianPugh, 2020-06-28:
[according to the VFS documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#_CPPv416esp_vfs_registerPKcPK9esp_vfs_tPv), `"/"` is an invalid base_path.

Still figuring out what to do/what is implied that might not be properly documents by the VFS code-base.

lorol, 2020-06-28:
Ok, i just forced this to become "" :)

lorol, 2020-06-28:
For sketches I am using actually the mounting point doesn't matter at all.

BrianPugh, 2020-06-28:
I'm gonna mess around with this a little more and report what I find here.

lorol, 2020-06-28:
Ok, you may check with this developer too:
https://github.com/earlephilhower/arduino-esp8266littlefs-plugin/issues/9

BrianPugh, 2020-06-28:
@lorol are you sure you're able to use `"/"` as a basepath? I'm unable to do this with SPIFFS (it fails the same logic).

In a similar note, I opened[ this issue](https://github.com/espressif/esp-idf/issues/5514) on the esp-idf repo questioning the path validation logic.

lorol, 2020-06-28:
I will check maybe I messed up the tings with ESP8266

BrianPugh, 2020-06-28:
[Igrr confirmed](https://github.com/espressif/esp-idf/issues/5514#issuecomment-650828458) that `"/"` is not a valid mounting point, so I'll mark this as closed for now. Thanks!

closed 2020-06-28 by BrianPugh

lorol, 2020-06-29:
Yes, you are right it was my mistake.