joltwallet/esp_littlefs #14: [Suggestion] #define CONFIG_LITTLEFS_SPIFFS_COMPAT

lorol, 2020-09-10:
@BrianPugh  Hi Brian, 
I believe 99%+ of esp32 applications use/test LittleFS now are based on previous SPIFFS code.  I am sure that in future more and more questions will appear about famous folder's differences, some will be breaking their app. until get fixed by iterating on higher level.
So my idea is, at your wrapper to be implemented say something like a config / #define CONFIG_LITTLEFS_SPIFFS_COMPAT , false by default; when enabled, emulates back the non-folder behavior of SPIFFS and people would activate it if desperately needed.
In fact, on esp8266 Arduino implementation it is already done this way by default, see here:
    - [recursive folders creation](https://github.com/esp8266/Arduino/blob/master/libraries/LittleFS/src/LittleFS.cpp#L60) when a new file is created to non-existing path
    - [recursive folders deletion](https://github.com/esp8266/Arduino/blob/master/libraries/LittleFS/src/LittleFS.h#L149) on "last file" deletion (SPIFFS cannot have "folder w/o file")

Could you please think of similar ideas at your driver level? Again, it can be conditional by config.
It looks like all dir operations needs to be reviewed in this regard like opendir() rmdir() mkdir() - big job.

I am sure, once LitleFS gets popular on esp32 all these questions will come anyway. Better to be prepared :) 
Appreciate your consideration.

BrianPugh, 2020-09-10:
sounds like a good idea, i might be able to tackle this this weekend.

lorol, 2020-09-10:
Really Great! 

Last example, I replaced with LittleFS in one interesting esp8266/esp32 project (FYI) https://github.com/rjwats/esp8266-react/issues/197
It is on PlatformIO esp32 arduino - they use old IDF 3.2 still. ...  
Almost 1:1 working, except few things that I need to tweak, they were OK on the ESP8266' LittleFS, which official driver was done intentionally "folderless-SPIFFS-compatible" style as I pointed.

On esp32 we don't have to default blindly this FS limitation to make it "the same", but if such choice is available by a config /#define, it will be awesome.
Have a nice weekend. 
BTW, I submitted to [Arduino Library Manager](https://github.com/arduino/Arduino/issues/10719) while IDF guys move on your PR :) and go forward,

BrianPugh, 2020-10-15:
@lorol just letting you know that i haven't forgotten about this, just other projects have been taking up my time. I'll try and get to this!

lorol, 2020-10-15:
@BrianPugh no worries. This  a low-priority. 
BTW, I see pwrite / pread additions but at Arduino core side, it seems to need other code to be "aware" of them, I guess at FS. 
Holding w/o these for now. The rest works fine.

BrianPugh, 2020-11-10:
alright @lorol I'm finally revisiting this. Should have some stuff in a bit. Ping me if any of these needs changes.

BrianPugh, 2020-11-10:
@lorol check out https://github.com/joltwallet/esp_littlefs/pull/20 and see if it meets the needs that you have observed.

lorol, 2020-11-10:
Sure, I will check. I probably will do a "last" manual update on my detached Arduino library and then all will be available through Arduino core by built-in EDF files.

lorol, 2020-11-10:
Hi @BrianPugh 
I did a "manual merge" test and so far it works fine. 
FYI, see how the esp8266 team "justified" this ugly thing (if someone asks later :)  why we provide such choice):
***"Subdirectories are automatically created when you attempt to create a file in a subdirectory, and when the last file in a subdirectory is removed the subdirectory itself is automatically deleted. This is because there was no mkdir() method in the existing SPIFFS filesystem."***
https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#littlefs-file-system-limitations

Still I recommend to be disabled by default in mainstream and later embedding to Arduino core, as you already did,

BrianPugh, 2020-11-10:
cool, i'll merge this soon

lorol, 2020-11-10:
Yes, whenever you decide.

BrianPugh, 2020-11-11:
merged

closed 2020-11-11 by BrianPugh

lorol, 2020-11-11:
👍