joltwallet/esp_littlefs #17: Does the littlefs support directories?

zht9961020, 2020-10-14:
Hi, does the littlefs support directories, or use a flat structure like spiffs?

lorol, 2020-10-14:
Yes, it does, unlike SPIFFS. 
You can create and remove directories.

zht9961020, 2020-10-14:
Another question, I create a file and write tens to hundreds of bytes to it, and close it. Then create another file and do this again. Then another file again. I found that the speed is very slow, why?

lorol, 2020-10-14:
Please provide numbers and comparison with SPIFFS as well as your expectations of "slow" and "fast".
Posting your code will help, too. Maybe your way of writing bytes involves longer processes and could be optimized.

zht9961020, 2020-10-16:
printf("-------------------------file system init use littlefs---------------------------\r\n");
    esp_vfs_littlefs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = "storage",//is not allowed to be NULL. You must specify the partition name from your partition table. 
    //   .max_files = 100,//field doesn't exist since we removed the file limit
      .format_if_mount_failed = true
    };
    esp_err_t ret = esp_vfs_littlefs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf("Failed to mount or format filesystem\r\n");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            printf("Failed to find SPIFFS partition\r\n");
        } else {
            printf("Failed to initialize SPIFFS (%d)\r\n", ret);
        }
        return;
    }
    size_t total = 0, used = 0;
    ret = esp_littlefs_info("storage", &total, &used);
    if (ret != ESP_OK) {
        printf( "Failed to get SPIFFS partition information\r\n");
    } else {
        printf( "Partition size: total: %d, used: %d\r\n", total, used);
    }

    //++++++++++++++++++++++++++++++++++++++++++++++
    //do some test
    printf("test littlefs write file start\r\n");
    FILE* FP;
    char file_name[100]={0};
    char str[1000] = "2222222222222222222222222222222222222211111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111";
    uint32_t len=0;

    for(int i=0; i<100; i++)
    {
        
        memset(file_name, 0, sizeof(file_name));
        sprintf(file_name,"/spiffs/a%d.txt", i);
        FP = fopen( file_name, "w" );
        fwrite(str, sizeof(str) , 1, FP );
        // f_write(FP, str, sizeof(str), &len);
        fclose(FP);

        printf("write file %s done\r\n", file_name);
    }
    printf("test littlefs write file done\r\n");
    //++++++++++++++++++++++++++++++++++++++++++++++

lorol, 2020-10-16:
IMO with this code you reopen and re/over write 100 times the entire file in your for() cycle, not very efficient way,
The underlying process involves buffering, checking and physical commit every time and that's why you "feel" it slow.
BTW, did you test with SPIFFS and compare, and does your code run faster with it?
Maybe @BrianPugh may elaborate more on this.

BrianPugh, 2020-10-16:
@zht9961020 can you post the results of your experiment? And as lorol said, depending on what you're trying to accomplish, the provided code is going to be quite slow on many systems. Your experiment is more about opening/closing file descriptors rather than actual read/write throughput.

zht9961020, 2020-10-17:
@BrianPugh I have hundreds of small files to save data. Each file size is  tens to hundreds of bytes, some larger files are 1k~10k.  
The application downloads these files from the network and updates them every few days.

BrianPugh, 2020-10-17:
so how slow is "slow"? what kind of speeds are you expecting? is another library offering higher throughput?

zht9961020, 2020-10-17:
#if USE_SPIFFS
    printf("-------------------------file system init use spiffs---------------------------\r\n");
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",//文件系统的目录地址
      .partition_label = NULL,//在.csv文件中的标签，如果设置为NULL则使用spiffs
      .max_files = 100,//同时可以打开最大的文件数
      .format_if_mount_failed = true//挂载失败就格式化
    };
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf("Failed to mount or format filesystem\r\n");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            printf("Failed to find SPIFFS partition\r\n");
        } else {
            printf("Failed to initialize SPIFFS (%d)\r\n", ret);
        }
        return;
    }
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        printf( "Failed to get SPIFFS partition information\r\n");
    } else {
        printf( "Partition size: total: %d, used: %d\r\n", total, used);
    }

    //++++++++++++++++++++++++++++++++++++++++++++++
    //do some test
    printf("test spiffs write file start\r\n");
    FILE* FP;
    char file_name[100]={0};
    char str[1000] = "2222222222222222222222222222222222222211111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111";
    uint32_t len=0;
    uint32_t t1=0;
    uint32_t t2=0;

    for(int i=0; i<100; i++)
    {
        
        memset(file_name, 0, sizeof(file_name));
        sprintf(file_name,"/spiffs/a%d.txt", i);
        t1=clock();
        FP = fopen( file_name, "w" );
        fwrite(str, sizeof(str) , 1, FP );
        // f_write(FP, str, sizeof(str), &len);
        fclose(FP);
        t2=clock();
        printf("write file %s done, time = %d ms\r\n", file_name, t2-t1);
    }
    printf("test littlefs write file done\r\n");
    //++++++++++++++++++++++++++++++++++++++++++++++
#endif

#if USE_LITTLEFS
    printf("-------------------------file system init use littlefs---------------------------\r\n");
    esp_vfs_littlefs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = "storage",//is not allowed to be NULL. You must specify the partition name from your partition table. 
    //   .max_files = 100,//field doesn't exist since we removed the file limit
      .format_if_mount_failed = true
    };
    esp_err_t ret = esp_vfs_littlefs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf("Failed to mount or format filesystem\r\n");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            printf("Failed to find SPIFFS partition\r\n");
        } else {
            printf("Failed to initialize SPIFFS (%d)\r\n", ret);
        }
        return;
    }
    size_t total = 0, used = 0;
    ret = esp_littlefs_info("storage", &total, &used);
    if (ret != ESP_OK) {
        printf( "Failed to get SPIFFS partition information\r\n");
    } else {
        printf( "Partition size: total: %d, used: %d\r\n", total, used);
    }

    //++++++++++++++++++++++++++++++++++++++++++++++
    //do some test
    printf("test littlefs write file start\r\n");
    FILE* FP;
    char file_name[100]={0};
    char str[1000] = "2222222222222222222222222222222222222211111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111";
    uint32_t len=0;
    uint32_t t1=0;
    uint32_t t2=0;

    for(int i=0; i<100; i++)
    {
        
        memset(file_name, 0, sizeof(file_name));
        sprintf(file_name,"/spiffs/a%d.txt", i);
        t1=clock();
        FP = fopen( file_name, "w" );
        fwrite(str, sizeof(str) , 1, FP );
        // f_write(FP, str, sizeof(str), &len);
        fclose(FP);
        t2=clock();
        printf("write file %s done, time = %d ms\r\n", file_name, t2-t1);
    }
    printf("test littlefs write file done\r\n");
    //++++++++++++++++++++++++++++++++++++++++++++++

#endif

zht9961020, 2020-10-17:
#if USE_SPIFFS
    printf("-------------------------file system init use spiffs---------------------------\r\n");
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 100,//
      .format_if_mount_failed = true
    };
    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf("Failed to mount or format filesystem\r\n");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            printf("Failed to find SPIFFS partition\r\n");
        } else {
            printf("Failed to initialize SPIFFS (%d)\r\n", ret);
        }
        return;
    }
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(NULL, &total, &used);
    if (ret != ESP_OK) {
        printf( "Failed to get SPIFFS partition information\r\n");
    } else {
        printf( "Partition size: total: %d, used: %d\r\n", total, used);
    }

    //++++++++++++++++++++++++++++++++++++++++++++++
    //do some test
    printf("test spiffs write file start\r\n");
    FILE* FP;
    char file_name[100]={0};
    char str[1000] = "2222222222222222222222222222222222222211111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111";
    uint32_t len=0;
    uint32_t t1=0;
    uint32_t t2=0;

    for(int i=0; i<100; i++)
    {
        
        memset(file_name, 0, sizeof(file_name));
        sprintf(file_name,"/spiffs/a%d.txt", i);
        t1=clock();
        FP = fopen( file_name, "w" );
        fwrite(str, sizeof(str) , 1, FP );
        // f_write(FP, str, sizeof(str), &len);
        fclose(FP);
        t2=clock();
        printf("write file %s done, time = %d ms\r\n", file_name, t2-t1);
    }
    printf("test littlefs write file done\r\n");
    //++++++++++++++++++++++++++++++++++++++++++++++
#endif

#if USE_LITTLEFS
    printf("-------------------------file system init use littlefs---------------------------\r\n");
    esp_vfs_littlefs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = "storage",//is not allowed to be NULL. You must specify the partition name from your partition table. 
    //   .max_files = 100,//field doesn't exist since we removed the file limit
      .format_if_mount_failed = true
    };
    esp_err_t ret = esp_vfs_littlefs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            printf("Failed to mount or format filesystem\r\n");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            printf("Failed to find SPIFFS partition\r\n");
        } else {
            printf("Failed to initialize SPIFFS (%d)\r\n", ret);
        }
        return;
    }
    size_t total = 0, used = 0;
    ret = esp_littlefs_info("storage", &total, &used);
    if (ret != ESP_OK) {
        printf( "Failed to get SPIFFS partition information\r\n");
    } else {
        printf( "Partition size: total: %d, used: %d\r\n", total, used);
    }

    //++++++++++++++++++++++++++++++++++++++++++++++
    //do some test
    printf("test littlefs write file start\r\n");
    FILE* FP;
    char file_name[100]={0};
    char str[1000] = "2222222222222222222222222222222222222211111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111";
    uint32_t len=0;
    uint32_t t1=0;
    uint32_t t2=0;

    for(int i=0; i<100; i++)
    {
        
        memset(file_name, 0, sizeof(file_name));
        sprintf(file_name,"/spiffs/a%d.txt", i);
        t1=clock();
        FP = fopen( file_name, "w" );
        fwrite(str, sizeof(str) , 1, FP );
        // f_write(FP, str, sizeof(str), &len);
        fclose(FP);
        t2=clock();
        printf("write file %s done, time = %d ms\r\n", file_name, t2-t1);
    }
    printf("test littlefs write file done\r\n");
    //++++++++++++++++++++++++++++++++++++++++++++++

#endif

zht9961020, 2020-10-17:
Use spiffs, running for the first time.
Log:
-------------------------file system init use spiffs---------------------------
W (6140) SPIFFS: mount failed, -10025. formatting...
Partition size: total: 956561, used: 0
test spiffs write file start
write file /spiffs/a0.txt done, time = 162 ms
write file /spiffs/a1.txt done, time = 161 ms
write file /spiffs/a2.txt done, time = 162 ms
write file /spiffs/a3.txt done, time = 161 ms
write file /spiffs/a4.txt done, time = 161 ms
write file /spiffs/a5.txt done, time = 161 ms
write file /spiffs/a6.txt done, time = 162 ms
write file /spiffs/a7.txt done, time = 162 ms
write file /spiffs/a8.txt done, time = 162 ms
write file /spiffs/a9.txt done, time = 162 ms
write file /spiffs/a10.txt done, time = 162 ms
write file /spiffs/a11.txt done, time = 162 ms
write file /spiffs/a12.txt done, time = 163 ms
write file /spiffs/a13.txt done, time = 163 ms
write file /spiffs/a14.txt done, time = 163 ms
write file /spiffs/a15.txt done, time = 163 ms
write file /spiffs/a16.txt done, time = 163 ms
write file /spiffs/a17.txt done, time = 162 ms
write file /spiffs/a18.txt done, time = 163 ms
write file /spiffs/a19.txt done, time = 164 ms
write file /spiffs/a20.txt done, time = 163 ms
write file /spiffs/a21.txt done, time = 163 ms
write file /spiffs/a22.txt done, time = 163 ms
write file /spiffs/a23.txt done, time = 164 ms
write file /spiffs/a24.txt done, time = 163 ms
write file /spiffs/a25.txt done, time = 164 ms
write file /spiffs/a26.txt done, time = 164 ms
write file /spiffs/a27.txt done, time = 164 ms
write file /spiffs/a28.txt done, time = 163 ms
write file /spiffs/a29.txt done, time = 164 ms
write file /spiffs/a30.txt done, time = 164 ms
write file /spiffs/a31.txt done, time = 164 ms
write file /spiffs/a32.txt done, time = 165 ms
write file /spiffs/a33.txt done, time = 164 ms
write file /spiffs/a34.txt done, time = 164 ms
write file /spiffs/a35.txt done, time = 164 ms
write file /spiffs/a36.txt done, time = 165 ms
write file /spiffs/a37.txt done, time = 165 ms
write file /spiffs/a38.txt done, time = 165 ms
write file /spiffs/a39.txt done, time = 164 ms
write file /spiffs/a40.txt done, time = 165 ms
write file /spiffs/a41.txt done, time = 164 ms
write file /spiffs/a42.txt done, time = 165 ms
write file /spiffs/a43.txt done, time = 165 ms
write file /spiffs/a44.txt done, time = 166 ms
write file /spiffs/a45.txt done, time = 165 ms
write file /spiffs/a46.txt done, time = 165 ms
write file /spiffs/a47.txt done, time = 166 ms
write file /spiffs/a48.txt done, time = 166 ms
write file /spiffs/a49.txt done, time = 166 ms
write file /spiffs/a50.txt done, time = 165 ms
write file /spiffs/a51.txt done, time = 166 ms
write file /spiffs/a52.txt done, time = 166 ms
write file /spiffs/a53.txt done, time = 167 ms
write file /spiffs/a54.txt done, time = 166 ms
write file /spiffs/a55.txt done, time = 166 ms
write file /spiffs/a56.txt done, time = 166 ms
write file /spiffs/a57.txt done, time = 167 ms
write file /spiffs/a58.txt done, time = 166 ms
write file /spiffs/a59.txt done, time = 166 ms
write file /spiffs/a60.txt done, time = 167 ms
write file /spiffs/a61.txt done, time = 167 ms
write file /spiffs/a62.txt done, time = 167 ms
write file /spiffs/a63.txt done, time = 167 ms
[... 119 lines cut ...]
write file /spiffs/a76.txt done, time = 73 ms
write file /spiffs/a77.txt done, time = 74 ms
write file /spiffs/a78.txt done, time = 74 ms
write file /spiffs/a79.txt done, time = 74 ms
write file /spiffs/a80.txt done, time = 74 ms
write file /spiffs/a81.txt done, time = 74 ms
write file /spiffs/a82.txt done, time = 74 ms
write file /spiffs/a83.txt done, time = 74 ms
write file /spiffs/a84.txt done, time = 74 ms
write file /spiffs/a85.txt done, time = 73 ms
write file /spiffs/a86.txt done, time = 73 ms
write file /spiffs/a87.txt done, time = 74 ms
write file /spiffs/a88.txt done, time = 74 ms
write file /spiffs/a89.txt done, time = 74 ms
write file /spiffs/a90.txt done, time = 74 ms
write file /spiffs/a91.txt done, time = 73 ms
write file /spiffs/a92.txt done, time = 73 ms
write file /spiffs/a93.txt done, time = 73 ms
write file /spiffs/a94.txt done, time = 74 ms
write file /spiffs/a95.txt done, time = 74 ms
write file /spiffs/a96.txt done, time = 74 ms
write file /spiffs/a97.txt done, time = 74 ms
write file /spiffs/a98.txt done, time = 74 ms
write file /spiffs/a99.txt done, time = 74 ms
test spiffs write file done

zht9961020, 2020-10-17:
Use spiffs, running for the first time.
Log:
-------------------------file system init use littlefs---------------------------
/components/esp_littlefs/src/littlefs/lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
W (6146) esp_littlefs: mount failed,  (-84). formatting...
Partition size: total: 1048576, used: 8192
test littlefs write file start
write file /spiffs/a0.txt done, time = 89 ms
write file /spiffs/a1.txt done, time = 89 ms
write file /spiffs/a2.txt done, time = 89 ms
write file /spiffs/a3.txt done, time = 88 ms
write file /spiffs/a4.txt done, time = 90 ms
write file /spiffs/a5.txt done, time = 90 ms
write file /spiffs/a6.txt done, time = 90 ms
write file /spiffs/a7.txt done, time = 88 ms
write file /spiffs/a8.txt done, time = 89 ms
write file /spiffs/a9.txt done, time = 92 ms
write file /spiffs/a10.txt done, time = 197 ms
write file /spiffs/a11.txt done, time = 88 ms
write file /spiffs/a12.txt done, time = 88 ms
write file /spiffs/a13.txt done, time = 90 ms
write file /spiffs/a14.txt done, time = 89 ms
write file /spiffs/a15.txt done, time = 90 ms
write file /spiffs/a16.txt done, time = 93 ms
write file /spiffs/a17.txt done, time = 91 ms
write file /spiffs/a18.txt done, time = 90 ms
write file /spiffs/a19.txt done, time = 90 ms
write file /spiffs/a20.txt done, time = 689 ms
write file /spiffs/a21.txt done, time = 89 ms
write file /spiffs/a22.txt done, time = 91 ms
write file /spiffs/a23.txt done, time = 92 ms
write file /spiffs/a24.txt done, time = 91 ms
write file /spiffs/a25.txt done, time = 91 ms
write file /spiffs/a26.txt done, time = 92 ms
write file /spiffs/a27.txt done, time = 93 ms
write file /spiffs/a28.txt done, time = 93 ms
write file /spiffs/a29.txt done, time = 966 ms
write file /spiffs/a30.txt done, time = 91 ms
write file /spiffs/a31.txt done, time = 92 ms
write file /spiffs/a32.txt done, time = 91 ms
write file /spiffs/a33.txt done, time = 93 ms
write file /spiffs/a34.txt done, time = 92 ms
write file /spiffs/a35.txt done, time = 93 ms
write file /spiffs/a36.txt done, time = 93 ms
write file /spiffs/a37.txt done, time = 93 ms
write file /spiffs/a38.txt done, time = 195 ms
write file /spiffs/a39.txt done, time = 91 ms
write file /spiffs/a40.txt done, time = 94 ms
write file /spiffs/a41.txt done, time = 93 ms
write file /spiffs/a42.txt done, time = 94 ms
write file /spiffs/a43.txt done, time = 93 ms
write file /spiffs/a44.txt done, time = 94 ms
write file /spiffs/a45.txt done, time = 1399 ms
write file /spiffs/a46.txt done, time = 92 ms
write file /spiffs/a47.txt done, time = 93 ms
write file /spiffs/a48.txt done, time = 92 ms
write file /spiffs/a49.txt done, time = 94 ms
write file /spiffs/a50.txt done, time = 96 ms
write file /spiffs/a51.txt done, time = 93 ms
write file /spiffs/a52.txt done, time = 1572 ms
write file /spiffs/a53.txt done, time = 94 ms
write file /spiffs/a54.txt done, time = 93 ms
write file /spiffs/a55.txt done, time = 95 ms
write file /spiffs/a56.txt done, time = 93 ms
write file /spiffs/a57.txt done, time = 96 ms
write file /spiffs/a58.txt done, time = 95 ms
write file /spiffs/a59.txt done, time = 1710 ms
write file /spiffs/a60.txt done, time = 94 ms
write file /spiffs/a61.txt done, time = 96 ms
write file /spiffs/a62.txt done, time = 95 ms
[... 121 lines cut ...]
write file /spiffs/a76.txt done, time = 105 ms
write file /spiffs/a77.txt done, time = 1181 ms
write file /spiffs/a78.txt done, time = 110 ms
write file /spiffs/a79.txt done, time = 113 ms
write file /spiffs/a80.txt done, time = 111 ms
write file /spiffs/a81.txt done, time = 111 ms
write file /spiffs/a82.txt done, time = 108 ms
write file /spiffs/a83.txt done, time = 106 ms
write file /spiffs/a84.txt done, time = 109 ms
write file /spiffs/a85.txt done, time = 108 ms
write file /spiffs/a86.txt done, time = 109 ms
write file /spiffs/a87.txt done, time = 109 ms
write file /spiffs/a88.txt done, time = 110 ms
write file /spiffs/a89.txt done, time = 1187 ms
write file /spiffs/a90.txt done, time = 105 ms
write file /spiffs/a91.txt done, time = 108 ms
write file /spiffs/a92.txt done, time = 104 ms
write file /spiffs/a93.txt done, time = 104 ms
write file /spiffs/a94.txt done, time = 105 ms
write file /spiffs/a95.txt done, time = 105 ms
write file /spiffs/a96.txt done, time = 105 ms
write file /spiffs/a97.txt done, time = 106 ms
write file /spiffs/a98.txt done, time = 105 ms
write file /spiffs/a99.txt done, time = 103 ms
test littlefs write file done

BrianPugh, 2020-10-17:
so to be clear, "very slow" in this situation is:

* ~1.8x faster than SPIFFS on the first write
* maybe about 30% slower on second write, which i suspect is mostly dominated by file descriptor acquiring/releasing?

also through all of this, you wrote <15KB of data, which hardly shows the limitations of SPIFFS

zht9961020, 2020-10-17:
littlefs first running:
"very slow", I mean this log, for all the 100 files, the whole time is much longer. Why these files take much longer time to write.
write file /spiffs/a10.txt done, time = 197 ms
write file /spiffs/a20.txt done, time = 689 ms
write file /spiffs/a29.txt done, time = 966 ms
write file /spiffs/a38.txt done, time = 195 ms
write file /spiffs/a45.txt done, time = 1399 ms
write file /spiffs/a52.txt done, time = 1572 ms
write file /spiffs/a59.txt done, time = 1710 ms
write file /spiffs/a65.txt done, time = 3543 ms
write file /spiffs/a80.txt done, time = 1892 ms
write file /spiffs/a87.txt done, time = 1642 ms
write file /spiffs/a99.txt done, time = 4458 ms

lorol, 2020-10-17:
Why these files take much longer time to write? - Due to open/close time because LittleFS does much more things to make FS robust against corruption.
You need to choose between robustness + extra milliseconds vs. faster open/close + chances to corrupt. Your choice.

zht9961020, 2020-10-17:
@lorol 
Spiffs and FatFS have their own problems. 
I haven't found a perfect solution to my problem.

Thank you very much!

closed 2020-10-19 by BrianPugh