joltwallet/esp_littlefs #23: README.md says to just clone esp_littlefs ... but neglects the need to clone src/littlefs submodule.

norlesh, 2020-11-22:
When esp_littlefs submodule is cloned using the suggested method in the README.md,  src/littlefs ends up as an empty directory... eventually got this working (am new to git) by first removing the littlefs and then adding the submodule in its place. On further research the answer [here on stackoverflow] suggests it should be as simple as adding `git submodule update --init --recursive` after esp_littlefs is added (I haven't tested this though).

(https://stackoverflow.com/questions/4600835/adding-git-submodule-that-contains-another-submodule)

BrianPugh, 2020-12-14:
i can update the readme to recursive clone

BrianPugh, 2020-12-14:
Added in https://github.com/joltwallet/esp_littlefs/commit/3152783bed1ce5d19dd64ddd098096d6fb9ecee8

closed 2020-12-14 by BrianPugh