joltwallet/esp_littlefs #24: Readme.md requires a short example of implementation in a project

wreyford, 2020-12-15:
It would be really helpful to new entry level users of your library, to have a basic example like the "hello world" for getting pointers on setting up everything correctly.
I set up a new project, using VSCode, with ESP-IDF extension (not Arduino), and not using PlatformIO.
I can build the lib, with the esp-littlefs lib in the components directory, but any calls to the esp-littlefs functions from my main_app fails.
I tried the example code in the littlefs originator lib you imported for esp, but that just wont work.

From the Espressif esp-idf examples, I can build and use the SPIFFS example without hassles.
Perhaps you can spend some time to help beginners like me to use your very needed package on first try.

BrianPugh, 2020-12-16:
so a lot of the unit tests are examples. The spiffs example should also just work if you change the `spiffs` part of the calls to `littlefs`. What kind of issues/errors are you having? linking?

wreyford, 2020-12-16:
Getting the CMakeLists.txt correct to reference the components properly, and the partition naming convention was part of my problem.
I took your advice and refactored the SPIFFS example from the ESP-IDF examples, and after many compiles and debugs eventually got it to work.
I decided to create a repo to document the task as I went, and perhaps it may be of value to others (saving a considerable amount of time for beginners like me).
Feel free to add it to your repo if you feel it adds value: https://github.com/wreyford/demo_esp_littlefs.git

BrianPugh, 2020-12-16:
would you be okay with me just linking your repo in the readme?

wreyford, 2020-12-16:
100%

BrianPugh, 2020-12-16:
Updated README, thanks!

closed 2020-12-16 by BrianPugh