joltwallet/esp_littlefs #26: LittleFS on external flash

tsctrl, 2021-01-11:
Hi @BrianPugh,

I successfully flash the internal flash of my esp32 bin file but not be able to flash external flash using the generated bin using mklittlefs. I am successfully doing the same using mkspiffs though.

`esptool.py --chip esp32 --port com6 --baud 115200 write_flash 0x0 ext-spiffs.bin --flash_mode dio --flash_size 16MB --spi-connection 18,19,23,21,5`
`esptool.py v3.1-dev`
`Serial port com6`
`Connecting........_`
`Chip is ESP32-D0WDQ6 (revision 1)`
`Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None`
`Crystal is 40MHz`
`MAC: 24:6f:22:c3:a3:54`
`Uploading stub...`
`Running stub...`
`Stub running...`
`Configuring SPI flash mode...`
`Configuring flash size...`
`Auto-detected Flash size: 16MB`
`Compressed 16777216 bytes to 499494...`
`Writing at 0x00004000... (6 %)`
`A fatal error occurred: Timed out waiting for packet header`

lorol, 2021-01-11:
Use esptool with --no-compress flag.
https://github.com/espressif/esptool/issues/577

tsctrl, 2021-01-11:
Hi @lorol,

Able to flash now, thank you!

closed 2021-01-11 by tsctrl