joltwallet/esp_littlefs #33: Automatic build and flash in esp-idf

IVData, 2021-06-19:
Hi,

Im having a hard time working out how to automatically build and flash a littlefs image using the esp-idf in vscode.

With spiffs you can do this by adding `spiffs_create_partition_image()` to cmake. I'd love a replacement for this using littlefs instead, but i can't find what this is actually running to be able to override it. 

Has anyone done this? 
Thanks

BrianPugh, 2021-06-23:
There's a similar tool here:

https://github.com/earlephilhower/mklittlefs

If you can add this functionality to this project's cmake, I'm totally open for PRs.

pyjamasam, 2021-07-06:
I haven't turned it into a PR yet, but https://github.com/pyjamasam/esp_littlefs/blob/master/project_include.cmake will allow you to add 
```
set(MKLITTLEFS "${PROJECT_DIR}/tools/mklittlefs/mklittlefs")
#littlefs_create_partition_image(spiffs-ui flash_data/ui FLASH_IN_PROJECT)
littlefs_create_partition_image(spiffs-ui flash_data/ui)
```
 To your CmakeLists.txt file in your project directory in an ESP-IDF 4.2 project.

There are some changes for 4.3 but my project uses 4.2, so I haven't sorted out what they are.

You need to still have mklittlefs somewhere and compiled (but you can specify the path as seen above).

IVData, 2021-07-06:
Awesome @pyjamasam , I will give this a go.

I have no cmake experience at all, does that file stay inside the LittleFS directory, or do I copy it to my project dir?

pyjamasam, 2021-07-06:
It stays in the esp_littlefs directory. I am using esp_littlefs as a component. So cmake seems to just pick it up automagically.

IVData, 2021-07-07:
Tested @pyjamasam 's solution and it works great. So I'll close this for now.

closed 2021-07-07 by IVData