joltwallet/esp_littlefs #34: sdcard mount?

maceip, 2021-07-01:
is it possible to mount an sdcard with something like  esp_vfs_littlefs_sdmmc_mount()? 

I attempted to hook this all up but from what I can tell esp_littlefs uses the partition table and I'm not sure how to get my sdcard to show up there...

BrianPugh, 2022-06-28:
There's PR #25 that apparently supports SD cards, but I haven't had the time to review such an extensive PR.

vbenso, 2023-03-11:
Hi @BrianPugh ,
Any news on this? Any luck with the mentioned PR?

BrianPugh, 2023-03-12:
I haven't really had any time to devote to adding new features to this project. I'm open to simple PRs, but I don't have resources to tackle something like SD card support. #25 may work for you, but I haven't personally played around with it.

redengin, 2023-07-23:
The current implementation is unnecessarily coupled to ESP32 flash where the ESP32 vfs provides full abstraction of the storage media. I'm happy to refactoring this code toward ESP32 a fully abstract usage of vfs.

BrianPugh, 2023-07-23:
@redengin can you please elaborate with examples? It's a bit hurtful, rude, and unhelpful to simply state "the implementation is full of unnecessary locking (and other naive mistakes)".

redengin, 2023-07-25:
I apologize for what I said. You've done a lot of valuable work.

I've now switched to FAT for my needs as they better support the user-usage.
If I have time, I'll fork and comment on some ways that the code-base can
be improved.

redengin, 2023-07-26:
I apologize for hurtful comments, they don't motivate progress - I'm
unnecessarily angry at the state of open source.

When I found your project I was looking for a littlefs implementation on
ESP-IDF.
I've done a lot of littelfs plumbing (gnu libc, newlib, musl). My two
cents, is that you should better understand the ESP vfs,
to plumb - rather than the specialization of spiff. The ESP-IDF
specializations for initializing  spiff partitions  (from a
local folder) are poorly designed and largely unnecessary (there are easier
ways to format and create a flash file).

BrianPugh, 2023-07-26:
I'm confused about what you are talking about, are you talking about `mklittlefs`, the external project that this uses to create premade littlefs images to flash to device? This github issue topic is for SD card support.

huming2207, 2024-01-09:
I will give it a try and add the support for SD card. I will see if I can draft a PR today or tomorrow.

BrianPugh, 2024-01-29:
@huming2207 has added SD card support with v1.13.0, so I'll close this issue for now. Thank you!

closed 2024-01-29 by BrianPugh