joltwallet/esp_littlefs #35: ftruncate() implementation missing

StridingDragon, 2021-07-28:
One of the key file handling functions, `ftruncate()` has not been implemented in this project, resulting in an incomplete file interface, even though there are clearly provisions for it on both the Espressif VFS end and LittleFS itself.

Is there chance to make the interface complete and get `ftruncate()` added?

StridingDragon, 2021-07-28:
I made the addition myself and in case you would like to add it to the project, here is the implementation for the truncate function. Naturally, it also requires the population of the `.truncate_p` element in the VFAS config table

```
        .truncate_p  = &vfs_littlefs_truncate,                  // $$$ Custom addition
```
…and this is the actual function…

```
static ssize_t vfs_littlefs_truncate( void *ctx, const char *path, off_t size )
{
    esp_littlefs_t * efs = (esp_littlefs_t *)ctx;
    ssize_t res = -1;
    vfs_littlefs_file_t *file = NULL;

    int fd = vfs_littlefs_open( ctx, path, LFS_O_RDWR, 438 );

    sem_take(efs);
    if((uint32_t)fd > efs->cache_size)
    {
        sem_give(efs);
        ESP_LOGE(TAG, "FD %d must be <%d.", fd, efs->cache_size);
        res = LFS_ERR_BADF;
    }
    else
    {
        file = efs->cache[fd];
        res = lfs_file_truncate( efs->fs, &file->file, size );
        sem_give(efs);

        if(res < 0)
        {
            errno = -res;
    #ifndef CONFIG_LITTLEFS_USE_ONLY_HASH
            ESP_LOGV(TAG, "Failed to truncate file \"%s\". Error %s (%d)",
                    file->path, esp_littlefs_errno(res), res);
    #else
            ESP_LOGV(TAG, "Failed to truncate FD %d. Error %s (%d)",
                    fd, esp_littlefs_errno(res), res);
    #endif
        }
        else
        {
            ESP_LOGV( TAG, "Truncated file %s to %u bytes", path, (uint32_t) size );
        }
    }
    vfs_littlefs_close( ctx, fd );
    return res;
}
```

BrianPugh, 2021-07-30:
Add in b1a6a623cb5127471021146c3259327182b60d2c. Thank you so much for your contribution!

closed 2021-07-30 by BrianPugh