joltwallet/esp_littlefs #36: error: conflicting declaration of '' with 'C' linkage

jogobeny, 2021-08-04:
When I am trying to `idf.py build` mentioned demo example (https://github.com/wreyford/demo_esp_littlefs) with `g++` compiler (`~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++`) I get this error message:
```
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:56:3: error: conflicting declaration of 'dirent* readdir(DIR*)' with 'C' linkage
   readdir(DIR *);
   ^~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:54:16: note: previous declaration with 'C++' linkage
 struct dirent* readdir(DIR* pdir);
                ^~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:58:6: error: conflicting declaration of 'int readdir_r(DIR*, dirent*, dirent**)' with 'C' linkage
 int  readdir_r(DIR *__restrict, struct dirent *__restrict,
      ^~~~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:59:5: note: previous declaration with 'C++' linkage
 int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);
     ^~~~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:61:7: error: conflicting declaration of 'void rewinddir(DIR*)' with 'C' linkage
 void  rewinddir(DIR *);
       ^~~~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:57:6: note: previous declaration with 'C++' linkage
 void rewinddir(DIR* pdir);
      ^~~~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:72:7: error: conflicting declaration of 'void seekdir(DIR*, long int)' with 'C' linkage
 void  seekdir(DIR *, long);
       ^~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:56:6: note: previous declaration with 'C++' linkage
 void seekdir(DIR* pdir, long loc);
      ^~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:73:7: error: conflicting declaration of 'long int telldir(DIR*)' with 'C' linkage
 long  telldir(DIR *);
       ^~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:55:6: note: previous declaration with 'C++' linkage
 long telldir(DIR* pdir);
      ^~~~~~~
In file included from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:76:6: error: conflicting declaration of 'int closedir(DIR*)' with 'C' linkage
 int  closedir(DIR *);
      ^~~~~~~~
In file included from ~/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/dirent.h:39,
                 from ~/Documents/app/components/esp_littlefs/include/esp_littlefs.h:17,
                 from ~/Documents/app/main/main.cpp:19:
~/Documents/app/esp-idf/components/newlib/platform_include/sys/dirent.h:58:5: note: previous declaration with 'C++' linkage
 int closedir(DIR* pdir);
     ^~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
I've done some research and including `sys/dirent.h` instead of `dirent.h` fixed my problem.

BrianPugh, 2021-08-06:
updated in d24949e, let me know if this solves your issue

closed 2021-09-14 by BrianPugh

jogobeny, 2021-09-30:
> updated in [d24949e](https://github.com/joltwallet/esp_littlefs/commit/d24949e7510e405f38de4821f6d8ac05ad3844b9), let me know if this solves your issue

Doing the same change in `esp_littlefs.h` solves my issue.