joltwallet/esp_littlefs #50: Unable to compile on S3, as not finding littlefs submodule, as not imported using component manager.

wreyford, 2021-12-17:
When building a project for ESP32 S3,  in Visual Studio Code Insiders on Windows 10, the component manager fetches the joltwallet__littlefs component, and adds it to the /managed_components folder.
The folder structure is correct, but for /littlefs ... see image below.
It appears not to import the submodule, so the /littlefs folder is empty...

[image]

So the build fails as it cannot find 
```
#include "littlefs/lfs.h"
```
Is there something I must add to .vscode settings.jason.
I have added:
```
"idf.enableIdfComponentManager": true,
```
Build error log follows

```
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: C:/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 3 dependencies:
[1/3] idf (4.4.0)
[2/3] joltwallet/littlefs (1.0.0)
[3/3] lvgl/lvgl (8.1.0)
CMake Error at C:/esp/esp-idf/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'esp_littlefs'.
Call Stack (most recent call first):
  C:/esp/esp-idf/tools/cmake/build.cmake:226 (__build_resolve_and_add_req)
  C:/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  C:/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:6 (project)
```

Please, any help will be appreciated.

BrianPugh, 2021-12-17:
Hmmm, so i've never actually used the component manager before, but downloading the archive directly from the website also has the missing files, so something is definitely not checking out the submodule somewhere. I'll see if I can debug this a lil.

BrianPugh, 2021-12-17:
alright @wreyford , I think I fixed it. I updated the CI so that it recursively checks out submodule when building the tarball for the IDF component manager. I also bumped the version so there's now v1.0.1.  I downloaded the tar from the official website and the files are there now, so hopefully this fixes your issue (lemme know!). Thanks for noticing and reporting this.

wreyford, 2021-12-17:
Thanks for the prompt reply @BrianPugh , unfortunately, still unable to compile.

```
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: C:/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 3 dependencies:
[1/3] idf (4.4.0)
[2/3] joltwallet/littlefs (1.0.1)
[3/3] lvgl/lvgl (8.1.0)
CMake Error at C:/esp/esp-idf/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'esp_littlefs'.
Call Stack (most recent call first):
  C:/esp/esp-idf/tools/cmake/build.cmake:226 (__build_resolve_and_add_req)
  C:/esp/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  C:/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "C:/ESP32/Projects/loadassist-s3/build/CMakeFiles/CMakeOutput.log".
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
The files appear to be in place, but perhaps its a missing #define somewhere for ESP32S3 that is resulting in it fialing to find the component in CMake.

Attached log file.
[CMakeOutput.log](https://github.com/joltwallet/esp_littlefs/files/7737074/CMakeOutput.log)

wreyford, 2021-12-17:
Perhaps altering from line 19 of esp_littlefs.c
```
#include <sys/param.h>
#if CONFIG_IDF_TARGET_ESP32C3
#include "esp32c3/rom/spi_flash.h"
#elif CONFIG_IDF_TARGET_ESP32S2
#include "esp32s2/rom/spi_flash.h"
#elif __has_include("esp32/rom/spi_flash.h")
#include "esp32/rom/spi_flash.h" //IDF 4
#else
#include "rom/spi_flash.h" //IDF 3
#endif
#include "esp_system.h"

#include "esp_littlefs.h"
#include "littlefs_api.h"
```
to:
```
#include <sys/param.h>
#if CONFIG_IDF_TARGET_ESP32S3
#include "esp32s3/rom/spi_flash.h"
#if CONFIG_IDF_TARGET_ESP32C3
```
 may help the compiler...

BrianPugh, 2021-12-17:
it looks like thats what spiffs is doing, are you able to test branches? I can push a branch with these fixes, but i don't have the devices.

wreyford, 2021-12-17:
I dont know if that is do-able with the component manager. Can give it a
try by cloning it in the normal way.

BrianPugh, 2021-12-17:
give #52 a try and lemme know how it goes.

wreyford, 2021-12-18:
I cloned in the repository using:
```
git clone -b greater-device-compat https://github.com/joltwallet/esp_littlefs.git components/esp_littlefs
```
The directory tree looks ok.
It is unable to compile with message:
```
In file included from ../components/esp_littlefs/src/esp_littlefs.c:41:
../components/esp_littlefs/include/esp_littlefs.h:21:10: fatal error: littlefs/lfs.h: No such file or directory
 #include "littlefs/lfs.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
```
Full error log:
```
> Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 .. <

-- Found Git: C:/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements
Updating lock file at C:/ESP32/Projects/loadassist-s3\dependencies.lock
Processing 2 dependencies:
[1/2] idf (4.4.0)
[2/2] lvgl/lvgl (8.1.0)
-- Project sdkconfig file C:/ESP32/Projects/loadassist-s3/sdkconfig
warning: The int symbol LV_USE_FS_STDIO (defined at C:/ESP32/Projects/loadassist-s3/managed_components/lvgl__lvgl/Kconfig:788) is being evaluated in a logical context somewhere. It will always evaluate to n.
warning: The int symbol LV_USE_FS_POSIX (defined at C:/ESP32/Projects/loadassist-s3/managed_components/lvgl__lvgl/Kconfig:795) is being evaluated in a logical context somewhere. It will always evaluate to n.
warning: The int symbol LV_USE_FS_WIN32 (defined at C:/ESP32/Projects/loadassist-s3/managed_components/lvgl__lvgl/Kconfig:802) is being evaluated in a logical context somewhere. It will always evaluate to n.
Loading defaults file C:/ESP32/Projects/loadassist-s3/sdkconfig.defaults...
-- Found PythonInterp: C:/esp/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") 
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "loadassist" version: 47ca40a-dirty
-- Adding linker script C:/ESP32/Projects/loadassist-s3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script C:/esp/esp-idf/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse escpos_printer esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_littlefs esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lvgl__lvgl lvgl_esp32_drivers lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/esp/esp-idf/components/app_trace C:/esp/esp-idf/components/app_update C:/esp/esp-idf/components/asio C:/esp/esp-idf/components/bootloader C:/esp/esp-idf/components/bootloader_support C:/esp/esp-idf/components/bt C:/esp/esp-idf/components/cbor 
C:/esp/esp-idf/components/cmock C:/esp/esp-idf/components/coap C:/esp/esp-idf/components/console C:/esp/esp-idf/components/cxx C:/esp/esp-idf/components/driver C:/esp/esp-idf/components/efuse C:/ESP32/Projects/loadassist-s3/components/escpos_printer C:/esp/esp-idf/components/esp-tls C:/esp/esp-idf/components/esp32s3 C:/esp/esp-idf/components/esp_adc_cal C:/esp/esp-idf/components/esp_common C:/esp/esp-idf/components/esp_eth C:/esp/esp-idf/components/esp_event C:/esp/esp-idf/components/esp_gdbstub C:/esp/esp-idf/components/esp_hid C:/esp/esp-idf/components/esp_http_client C:/esp/esp-idf/components/esp_http_server C:/esp/esp-idf/components/esp_https_ota C:/esp/esp-idf/components/esp_https_server C:/esp/esp-idf/components/esp_hw_support C:/esp/esp-idf/components/esp_ipc C:/esp/esp-idf/components/esp_lcd C:/ESP32/Projects/loadassist-s3/components/esp_littlefs C:/esp/esp-idf/components/esp_local_ctrl C:/esp/es [... cut ...]
../components/esp_littlefs/src/littlefs_api.c:12:10: fatal error: littlefs/lfs.h: No such file or directory
 #include "littlefs/lfs.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
[1036/1247] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/core/lv_indev_scroll.c.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

BrianPugh, 2021-12-18:
whenever you directly clone, you have to pass in the `--recursive` flag, so:
```
git clone -b greater-device-compat --recursive https://github.com/joltwallet/esp_littlefs.git components/esp_littlefs
```

Or, in an already cloned repo, the command to initialize/pull/checkout all the appropriate submodules is:
```
git submodule update --init --recursive
```

wreyford, 2021-12-18:
Awesome, thanks!

BrianPugh, 2021-12-18:
please lemme know if it works for you!

wreyford, 2021-12-19:
Thx a lot @BrianPugh,
The code compiles perfectly for the S3 build. I have sent the binary off to our hardware guy, and he will test his side.
I'm sure all should be fine.
Are you going to push the #define updates to your component, as I would prefer to use LittleFS as a component.

jrgilman, 2021-12-19:
> alright @wreyford , I think I fixed it. I updated the CI so that it recursively checks out submodule when building the tarball for the IDF component manager. I also bumped the version so there's now [...]

This just saved my butt, I had the exact same issue as @wreyford described. Should the README be updated to say to use 1.0.1? I can submit a PR if necessary.

wreyford, 2021-12-20:
@BrianPugh I confirm that LittleFs not only compiles for S3 but our preliminary tests indicate thst it is working as expected.
Thx a lot.

closed 2021-12-20 by wreyford

BrianPugh, 2021-12-20:
I released version 1.1.0 with these added devices. @jrgilman I updated the README.