joltwallet/esp_littlefs #54: Build error

shark8594, 2022-01-11:
Hi,

Tried to include esp_littlefs under componets and the build seems to fail, any clue? Error below:

FAILED: C:/Users/abhin/esp/components/esp_littlefs/mklittlefs/mklittlefs
cmd.exe /C "cd /D C:\Users\abhin\esp\components\esp_littlefs\mklittlefs && make dist"
FIND: Parameter format not correct
g++ -std=gnu++11 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -I ../src -D VERSION=\"0.2.3-40-ga14dabe\" -D LITTLEFS_VERSION=\"v2.4.1\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o main.o main.cpp
cc -std=gnu99 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -I ../src -D VERSION=\"0.2.3-40-ga14dabe\" -D LITTLEFS_VERSION=\"v2.4.1\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o ../src/littlefs/lfs.o ../src/littlefs/lfs.c
process_begin: CreateProcess(NULL, cc -std=gnu99 -Os -Wall -Itclap -Iinclude -Ilittlefs -I. -I ../src -D VERSION=\"0.2.3-40-ga14dabe\" -D LITTLEFS_VERSION=\"v2.4.1\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32 -c -o ../src/littlefs/lfs.o ../src/littlefs/lfs.c, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [../src/littlefs/lfs.o] Error 2

Thanks!

BrianPugh, 2022-01-12:
did you clone this repo? you have to add the `--recursive` command:

```
git clone --recursive https://github.com/joltwallet/esp_littlefs.git
```

or, in your already cloned repo, run:

```
git submodule update --init --recursive
```

shark8594, 2022-01-12:
Thanks for the quick response, the issue that fixed it was adding a symbolic link to gcc named cc.

closed 2022-01-12 by shark8594