joltwallet/esp_littlefs #57: undefined reference to `ftruncate'

Rajssss, 2022-01-23:
Is it possible to add ftruncate support, or anyway I can use truncate for the same effect?

BrianPugh, 2022-01-31:
i'm going to close this for now, we can reopen it once there's upstream support.

igrr, 2022-03-10:
Hi @BrianPugh, upstream support has been added in https://github.com/espressif/esp-idf/commit/ea9f7cc3705ab70736504f92f6100a1b721c1841. It will be released in IDF v5.0. We are also discussing backporting it to release/v4.4 branch and making it available in v4.4.2 bugfix release.

You can wrap `.ftruncate_p` line in `esp_vfs_t` structure initializer with IDF version check similar to this one:
```c++
#include "esp_idf_version.h"

// later in esp_vfs_littlefs_register
#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0)
    .ftruncate_p     = &vfs_littlefs_ftruncate,
#else
```

BrianPugh, 2022-03-10:
thanks @igrr , re-opening this issue, will try and add this soon

closed 2022-08-03 by BrianPugh