joltwallet/esp_littlefs #59: Concatenate files?

ndjur, 2022-01-27:
Hi,
I'm logging ESP32 debug output to a temporary log file and every time ESP32 starts, I would like to concatenate this temporary file to a "permanent" log file that can be later extracted by some host computer (no wifi connection, only BLE - sometimes). New temporary log file would be opened. 
I was hoping that doing the concatenation instead of coping and appending to a big log file would minimize flash wear as the location of the new temporary file would always be different. Does littleFS allows this and would it really help with the flash wear? Another option is to write directly to a "big" log, but in case this one gets corrupted everything is lost. How big is "big"? My board has WROVER with 16MB of flash and half of this is dedicated for debug log data.
I would also have an indexing file that keeps position of each temporary log file inside the big log.
Thanks.

BrianPugh, 2022-01-27:
If I were you, i'd write to several smaller (say 512KB or 1MB) log files and then concatenate when extracting to a host computer. You don't have to worry about the wear leveling, littlefs handles all that for you.

ndjur, 2022-01-27:
Thank you very much, that's what I'm doing now but for file naming I'm using a counter from which is saved in NVM and rewritten every time. I was just wondering if even this re-writing can be avoided. Should it?

ndjur, 2022-01-27:
Just one more question.
Every time the ESP32 application starts its debug output (the one that is normally outputted via serial output) is logged into a "temporary" log file.  Next start ESP32 starts this log file is renamed into a new name that is created based on the NVS previous session counter. The new "temporary" log is then started. These ESP32 sessions can be very short, so the log files can be very small. From littleFS point of view - is it better to log several sessions into a temporary log so that all logs are, like you say, 512K - 1MB, or it is OK to have many small files that are sometimes as small as 1K?
Thank you.

BrianPugh, 2022-01-31:
it's fine to have many small files, but i think you might be prematurely optimizing a bit.

closed 2022-03-29 by BrianPugh