joltwallet/esp_littlefs #61: std::fflush don't works

ilutchenko, 2022-02-28:
Hi Brian!

First, thank you for this port of LittleFS!

But I've faced some issues here.
I want to use LittleFS for errors logs storing.
I use esp_littlefs `v1.1.0` and [spdlog](https://github.com/gabime/spdlog) for files rotating. My log files are limited to 1KB size.
After writing each error message these files are flushed with `std::fflush`.

And I see that LittleFS does not save files to the flash.
It saves files only when the max size is reached and the file is closed.

There you can see that the current working logfile size is zero (messages are buffered but not written to the flash):
[image]

But it works fine with spiffs (small messages successfully written to the flash):
[image]

My point is that LittleFS does not transfer data to the flash when `std::fflush` is triggered.
I've tried to decrease `Minimum size of a block write`, but have had no success.
Can you make some suggestions how to fix it?

BrianPugh, 2022-02-28:
So i'll be honest, i never program in CPP. However, is this a misunderstanding between fflush and fsync, or do you think its an actual bug?

See related issue in esp-idf:
https://github.com/espressif/esp-idf/issues/2820#issuecomment-446051814

ilutchenko, 2022-03-04:
> is this a misunderstanding between fflush and fsync

Hmm, yes, looks like this is the root of the problem.
But it also means that the behavior of littlefs differs from the behavior of spiffs. And it breaks drip-in compatibility.

We can fix it by adding `vfs_littlefs_fsync()` into `vfs_littlefs_write()`.
What do you think about adding another one config line into `kconfig` and wrap `vfs_littlefs_fsync()` into `#ifdef`?
If this sounds good to you, I can prepare PR for it.

BrianPugh, 2022-03-08:
thanks for the PR!

closed 2022-03-08 by BrianPugh