joltwallet/esp_littlefs #64: Issue about return code

donghengqaz, 2022-03-14:
I see the code return the littlefs's error code to VFS, and VFS pass this code to application, for example the [code](https://github.com/joltwallet/esp_littlefs/blob/4ee5881509360d6c58b16bbc698218ee5e21b4d2/src/esp_littlefs.c#L1270).

BrianPugh, 2022-03-14:
what issues is this causing? if it's returning a nonzero value, then some error-handling action needs to be taken.

donghengqaz, 2022-03-15:
> what issues is this causing? if it's returning a nonzero value, then some error-handling action needs to be taken.

Yes, but some open source do like following:

```
int ret = fstat(fd, buf);
if (ret != -1) {
 ...
} else {
 ...
}
```

They just checking the result if it is `-1` instead of if it is `< 0`.

BrianPugh, 2022-03-15:
hmmm, ok I think you're right. According to all docs I can find, a lot of these functions are:

>On success, zero is returned. On error, -1 is returned, and errno is set appropriately.

I erroneously thought it was just "negative number on failure". I don't quite have the time right now to address this, but it'd be good to make this a kconfig option.

hmimberg, 2022-03-17:
When looking through the code I found, that in
https://github.com/joltwallet/esp_littlefs/blob/4ee5881509360d6c58b16bbc698218ee5e21b4d2/src/esp_littlefs.c#L152
the returned value of esp_littlefs_by_label is checked: 
 ```c++
err = esp_littlefs_by_label(partition_label, &index);
if(err != ESP_OK) return false;
```

I was wondering if this is by design. If I am not mistaken, ESP_OK == false, therefore this function returns ESP_OK.

BrianPugh, 2022-03-17:
@hmimberg I believe thats a bug; it should return the `err`. I'll try and start tackling some of these issues this weekend, but I can't promise i'll get around to it. PRs are always appreciated 😄

BrianPugh, 2022-03-29:
fixed the wrong `err` returned here:

https://github.com/joltwallet/esp_littlefs/pull/73/files

BrianPugh, 2022-03-29:
So i'm going to go through and change all the return values to -1 and set the errno appropriately. While this is backwards incompatible in some cases, I think its the right way to move forward. In some calls (such as `fseek`), the `vfs` layer was changing some negative return values to 0 (changing -22 to 0). So some stuff was already broken. Expect this to be resolved within the next few days.

BrianPugh, 2022-03-30:
Resolved in #75 . If this doesn't resolve your issues i'll reopen. I'll cut a release within the next few days.

closed 2022-03-30 by BrianPugh