joltwallet/esp_littlefs #65: Issues about fseek

donghengqaz, 2022-03-14:
The littlefs lacks of full method to checking the arguments of fseek when the target position is out of range, could you please add checking method in the VFS wrapper function [vfs_littlefs_lseek](https://github.com/joltwallet/esp_littlefs/blob/4ee5881509360d6c58b16bbc698218ee5e21b4d2/src/esp_littlefs.c#L1174).

donghengqaz, 2022-03-15:
> lfs should internally be detecting that here:
> 
> https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.c#L3075

Here it checks the file limitation not current file's size.

BrianPugh, 2022-03-15:
hm, I think you're right again. Yes, we should add some oob checks.

BrianPugh, 2022-03-29:
so [according to the man page](https://man7.org/linux/man-pages/man2/lseek.2.html), you're perfectly allowed to seek past the end of a file. I added unit tests to confirm it behaves as indicated.

Also, internally lfs actually checks if you attempt to seek to before-the-start of a file. However, turns out that me returning negative values besides `-1` was causing issues in the vfs layer.

So, the fix is to properly set the `errno` and return -1 (as you state to in #64). Resolved in #74.

closed 2022-03-29 by BrianPugh