joltwallet/esp_littlefs #69: It is better to move not necessary `#include`s in header files to source files.

7aman, 2022-03-28:
There are some `#include` directives in `esp_littlefs.h` which are not needed by the header itself. It seems in contrary with recommendations about header files. It pollutes source files that include this header file.

```C
#include <stdint.h>
#include <stddef.h>
#include <stdarg.h>
#include <unistd.h>
#include <utime.h>
#include "freertos/FreeRTOS.h"
#include "freertos/semphr.h"
#include "esp_err.h"
#include <sys/types.h>
#include <sys/reent.h>
#include <sys/stat.h>
#include <sys/time.h>
#include <sys/poll.h>
#include <sys/dirent.h>
#include <string.h>
#include "sdkconfig.h"

#include "littlefs/lfs.h"
```

could be reduced to

```C
#include "esp_err.h"
#include "littlefs/lfs.h"
#include "sdkconfig.h"
```
and other `#include`s should be in source files.

BrianPugh, 2022-03-28:
thanks! i'll address this later today; just trying to get CI working again first.

BrianPugh, 2022-03-29:
addressed in https://github.com/joltwallet/esp_littlefs/pull/71

closed 2022-03-29 by BrianPugh