joltwallet/esp_littlefs #76: Cannot rename and cannot not delete

rolanddemeester, 2022-04-01:
I tried your example and found out that renaming a file fails.
I changed the example a bit and tried to delete the created file: no luck.
then I used your updated version of esp_littlefs.c  to run the part of your test that deals with renaming: no improvement. Deleting fails here as well. 
When I used errno to get more info, I received each time "88", very strange.

BrianPugh, 2022-04-01:
interesting; does it still return 0 or does it now return -1 on failure?

rolanddemeester, 2022-04-02:
It returns -1. The test crashes with a “StoreProhibited”, but I don’t know if that’s relevant (it may be my set-up).

 

 

Test setup complete.

I (5244) demo_esp_littlefs: OK, name_src = /littlefs/move_src.txt, name_dst = /littlefs/move_dst.txt

:68::FAIL: Expected 0 Was -1Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

BrianPugh, 2022-04-04:
hmmm, i just re-ran the demo on my chip and am unable to reproduce.

rolanddemeester, 2022-04-19:
I had to install the latest stable version of ESP-IDF (4.4) for another project and used it to rebuild the demo.

Now it works: renaming and deleting.

closed 2022-04-19 by rolanddemeester

MathieuAP, 2022-04-28:
Hi,

I think that I have the same problem.

idf.py --version: **ESP-IDF v4.4-dirty**

_(ESP32-WROOM-32)_

My init():
```
    esp_vfs_littlefs_conf_t conf = {
        .base_path = "/littlefs",
        .partition_label = "littlefs",
        .format_if_mount_failed = true,
        .dont_mount = false};

    esp_err_t ret = esp_vfs_littlefs_register(&conf);
```

My attempt to delete a file
```
    std::string pathSpiff = "/littlefs/" + std::string(name);
    printf("Deleting file: %s\r\n", pathSpiff.c_str());

    // unlink
    int ret = unlink(pathSpiff.c_str());  -> return "-2" and no effect
    if (ret != 0) {
        printf("- unlink failed %d\n", ret);
    } else {
        printf("- file unlink\n");
    }

    // remove
    ret = remove(pathSpiff.c_str());  -> return "0" but no effect
    if (ret != 0) {
        printf("- remove failed %d\n", ret);
    } else {
        printf("- file remove\n");
    }

    // rename
    ret = rename(pathSpiff.c_str(), "/littlefs/foo.txt");  -> return "-2" and no effect
    if (ret != 0) {
        printf("- rename failed %d\n", ret);
    } else {
        printf("- file rename\n");
    }
```

BrianPugh, 2022-04-28:
Do you have the latest version of esp_littlefs?

MathieuAP, 2022-04-29:
Hi, 
I upgraded my esp_littlefs version from **1.0.0** to **1.1.0** and now, it works perfectly.

Thank you very much for your work!

BrianPugh, 2022-04-29:
@MathieuAP so I just reallized the version on the espressif component manager was our of date (v1.1.0 is quite old). I just uploaded the latest version `v1.3.1`. Can you update again and confirm everything still works for you?

MathieuAP, 2022-05-02:
@BrianPugh,

Thank you for the update. Everything is working with the **v1.3.1** yes 👍.

rolanddemeester, 2022-10-11:
Should we look at differences in the development environment?

 

I am using VSCode with the ESP-IDF plugin from Expressif.

I didn’t change anything in the LittleFS section of the SDKConfig.

I used the same partition layout as for the unit tests.

I used 2 ESP32 WROOM32D, one development board and a bare one to exclude the possibility of a faulty device.

I am puzzled by the ‘StoreProhibited’ exception, I don’t know if this is caused by the rename or by the test result.

Do you have any ideas how to proceed?