joltwallet/esp_littlefs #78: Inconsistency in `LFS_NAME_MAX` settings (LittleFS / mklittlefs)

Jason2866, 2022-04-24:
LittleFs uses as default `64` https://github.com/joltwallet/esp_littlefs/blob/master/Kconfig#L24
For mklittlefs the setting is `32` https://github.com/BrianPugh/mklittlefs/blob/a14dabe444e9a265aab04d085d8ca8f8536d799b/Makefile#L75
Which generates compability problems. ESP8266 uses `32` in general.
What is the reason for choosing `64` for the ESP32 IDF / Arduino implementation?

BrianPugh, 2022-04-24:
I think the choice was pretty arbitrary, but it should definitely be fixed to be consistent. Do you have a suggestion/justification for whether the default should be 32 or 64?

Jason2866, 2022-04-24:
Good question! It would be nice to be align with esp8266 BUT since LittleFS is introduced in IDF the default 64 is used to compile the Arduino Esp32 librarys for the framework. So a change would be a breaking change here.

BrianPugh, 2022-04-24:
so I don't really use arduino stuff, but is the following correct:
1. There's this repo, the "root" project for littlefs on the esp32.
2. There's[ lorol's repo,](https://github.com/lorol/LITTLEFS) which is sort of like a fork of this repo. This repo is only for arduino esp32 core v1.X.
3. Lorol's [repo got copied over into arduino-esp32](https://github.com/espressif/arduino-esp32/tree/master/libraries/LittleFS).

I'm assuming the `mklittlefs` you're referring to is [linked here](https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_PlatformIO/README.md), which links to earlephilhower/mklittlefs, the creator of `mklittlefs`. And those binaries were compiled using the 32-byte path length default.

So, with all of that in consideration, it seems like the easiest way forward is to change the default in this repo for consistency, **and** independently in the arduino-esp32 to 32 to resolve your issue. Does this seem correct? Because if so, you should also open an issue in `arduino-esp32`. I didn't see where this value gets configured in the other projects.

Jason2866, 2022-04-25:
Hello,
1. and 2. is correct. 3. This repo is used for IDF and Arduino  for LittleFS

BrianPugh, 2022-04-25:
so for this repo, I'd prefer to keep it to 64 so that it doesn't catch anyone off guard.

I opened up this PR that should pass the config value to the build step of mklittlefs

https://github.com/joltwallet/esp_littlefs/pull/79

Maybe arduino should do something similar instead of downloading a precompiled binary.

I'll merge it when i get a chance to test and think about it more.

Jason2866, 2022-04-25:
Thanks for heads up!

A6GH, 2022-12-15:
What is the meaning of LITTLEFS_OBJ_NAME_LEN parameter? What does it limit exactly? I've tried creating files with longer names than 64 and it works fine. Could someone explain?

aircable, 2023-04-24:
I had the same limit problem not being able to build a littlefs file system with the mklittlefs code within the IDF.
This is in conjunction with sveltekit building static web sites for the ESP32. Filenames are just longer.
I think the mklittlefs should get the configuration for the LFS_NAME_MAX from the sdkconfig CONFIG_LITTLEFS_OBJ_NAME_LEN when building the binary. Even though this is set to larger, mklittlefs will still fail to build the file system.

BrianPugh, 2023-05-03:
Fixed in #79. Will be a part of v1.5.4

closed 2023-05-03 by BrianPugh