joltwallet/esp_littlefs #80: Problem in real heavy use environemnt

filzek, 2022-05-04:
Hi,

We have give a chance to LITTLEFS after all currently loosing partition after poweroutgaes with file writtings happening, so, we decide to try the LITTLEFS in the currently fw we use on our devices, so far not good the LITTLEFS creates a lot fo problem in all around with the SDK 4.4.1.

IPC/WDT calls, afecting all around. 

SDK 4.4.1

LittleFS as suggest to replace FATFS or SPIFFS, but it has a lot of problems of IPC calls, hangs and concurrent executions.

The behavior in a complex environment is worthless, crosscore and make the tasks halt forever.

If you are accessing the flash you are dead, nothing work flawless or concurrent, it just crash the entire system.
Backtrace:0x4008FAD1:0x3FFBFCA00x40084301:0x3FFBFCC0 0x400D97CF:0x3FFEC380 0x4009AD45:0x3FFEC390 0x400827D1:0x3FFEC3B0 0x4008EAE3:0x3FFEC3D0 0x4008EDF8:0x3FFEC3F0 0x4008EEFE:0x3FFEC440 0x4014BBF9:0x3FFEC470 0x4014BC6D:0x3FFEC490 0x400FA689:0x3FFEC4C0 0x400FABAD:0x3FFEC4E0 0x400FAE2A:0x3FFEC510 0x400FAEBE:0x3FFEC580 0x400FC8E5:0x3FFEC5F0 0x400FDBAD:0x3FFEC6B0 0x400FA1B6:0x3FFEC7A0 0x400D53C5:0x3FFEC7C0 0x401BAF45:0x3FFEC7E0 0x400E1859:0x3FFEC800
0x4008fad1: esp_crosscore_isr at D:/Dropbox/esp-idf/components/esp_system/crosscore_int.c:92

0x40084301: _xt_lowint1 at D:/Dropbox/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

So, to replace currently using FAFTS/SPIFFS to LITTLEFS is not a working solution.

Something in the LITTLEFS makes the cpu not to yeld correct to another tasks and create lags and halts everywhere in the code.

BrianPugh, 2022-05-04:
this is a fairly light wrapper/port of littlefs for the esp32, so these are more issues that would either be need to be brought up with:

1. esp-idf if its something in how tasks are being scheduled or stuff like that.
2. upstream littlefs if its an actual filesystem issue.

However, if you can find/know of something in this port that is specifically implemented incorrectly or could be better, I'd be more than glad to help and/or merge PRs.

filzek, 2022-05-04:
Hi @BrianPugh, we have very intense tasks and interrupts, also we have interrupts with adc readings, but with FatFS or SPIFFS it seens to work perfect without any wdt / crosscore issues. It is very intense how the interrupts rush against each another, the most time consumption cpu is the adc_readings that almost take 80% of free time of the cpu per second, it does a timer and than a timer dispatch the handle to run it, so, it keeps like it forever.

It does use http task every second or 2 seconds.

Its only read large files once it has boot, so no task or anything intese is running.

But when it start to run, it only writes to the files whenever changes happen or the save to file has been called, so, its not wirte and read intense.

What make the things interesting is that with the lLittleFS running the system becomes unstable, so all errors come abroad.

This seems that whenever you have the LittleFS running (like a partition mounted) and the system is very busy it start to crash all around.

BrianPugh, 2022-05-04:
hmmm, maybe the other file systems have larger caches and thusly have reduced flash access? Try increasing `LITTLEFS_CACHE_SIZE` and see if that does anything. Also, do you `LITTLEFS_FLUSH_FILE_EVERY_WRITE` enabled? Because that will also increase flash access.

dalbert2, 2022-06-17:
For tasks that hit flash constantly, any file system may not be the right answer.  We use the flash filesystem to store infrequently used data but for persistent data that must be written quickly and often, we use circular logs maintained directly in their own flash partitions.  File systems aren't really meant for fast data logging.

BrianPugh, 2022-08-09:
There seems to be no actionable items here, so I'm going to close this issue for now. If more information comes available, we can reopen.

closed 2022-08-09 by BrianPugh