joltwallet/esp_littlefs #81: LittleFS on external flash of 128MB

thyclouds, 2022-06-15:
Hi!

We have reached great read numbers with LittleFS on internal flash, however we need more space for our scenario. We want to use LittleFS with Winbond [W25Q01JVZEIQ](https://eu.mouser.com/datasheet/2/949/W25Q01JV_SPI_RevB_11132019-1760416.pdf) SPI Flash which is a 1Gbit (128MB flash). We have soldered it to ESP32-S3 and are able to use the lower 16MB for internal flash, and now want to use it as a dedicated external SPI flash. Do you have any idea on how to use LittleFS to lay a partition of 128MB? 

A couple of thoughts that we had:
- Does it need to be split to 16 or 32MB partitions as described in this link [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/chip-series-comparison.html#id5)?
- Should we use approach described in #26 but with 128MB? 
- What we saw is that in order to utilize Flash above 16MB one has to change from 24-bit to 32-bit address space using an SPI command. I doubt that it would work automagically for LittleFS.

Greatly appreciate your thoughts and support!

BrianPugh, 2022-06-16:
Thanks for the report! Off the top of my head, I'm not totally sure why this limitation exists, I'll try and get you a more helpful response when I have more time this weekend to think about it and investigate. If you find any additional information that you think is relevant, please keep me updated! I don't have such a large flash chip to run tests, but might hopefully find some leads.

BrianPugh, 2022-06-21:
So I didn't do any diving into the code, but I don't think theres any reason why this port shouldn't work with larger chips. A brief amount of research says that the ESP32 is compatible with 32-bit addressing, it just uses 24-bit addressing by default because 32-bit addressing has [vendor specific commands](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_optional_feature.html#bit-address-flash-chips).

That said, this littlefs port builds upon the VFS in esp-idf, which uses the SPI Flash API. I think if you [override the default chip drivers](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_override_driver.html), everything downstream (the VFS and littlefs) should work automagically.

Let me know if you have any success/roadblocks with this!

BrianPugh, 2022-08-09:
going to close this, but if you have any updates, feel free to post back here.

closed 2022-08-09 by BrianPugh