joltwallet/esp_littlefs #82: base path '/'

dalbert2, 2022-06-17:
I'm using PlatformIO with VSCode and ESP-IDF framework. I've imported the LittleFS component and it works nicely, but I can't register LittleFS at the base path of "/" (I think it's because SPIFFS is registered there). I didn't see anything in menuconfig that would let me remove SPIFFS.

Is it possible to register LittleFS as the root file system

BrianPugh, 2022-06-17:
According to the esp-vfs docs, `"/"` is an invalid mount point, and an empty mount point should be used instead.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#paths

See this issue:
https://github.com/joltwallet/esp_littlefs/issues/11

Let me know if you have any issues with an empty mount point.

dalbert2, 2022-06-17:
Thanks Brian, that did the trick!
Also, may I ask how to enable mkdir() and rmdir() ?

BrianPugh, 2022-06-17:
If `CONFIG_VFS_SUPPORT_DIR` is enabled, it should all just work

 https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/kconfig.html#config-vfs-support-dir

dalbert2, 2022-06-17:
Thank you again, both for your help and for the awesome port of LittleFS!

These items are small, but the solutions are not obvious (until you already know them).  Adding things like these to the wiki will doubtless help others.

closed 2022-06-17 by dalbert2