joltwallet/esp_littlefs #83: littlefs_create_partition_image() in VSCode CMakeList.txt fails

jeeper1974, 2022-07-11:
I was able to get mklittlefs added to esp-idf 4.4 components and everything builds fine. When I add littlefs_create_partition_image() to CMakeLists.txt it fails. I can build the SPIFFS partition with spiffs_create_partition_image() without issue as shown here. Littlefs_create_partition_image() fails.

`#spiffs_create_partition_image(storage ../spiffsImage FLASH_IN_PROJECT)
set(MKLITTLEFS "C:/Users/roger/esp4.4B/esp-idf/components/esp_littlefs/mklittlefs/mklittlefs")
littlefs_create_partition_image(storage ../spiffsImage)`

`[2/6] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\bootloader\esp-idf\esptool_py && C:\Users\roger\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/roger/esp4.4B/esp-idf/components/partition_table/check_sizes.py --offset 0x9000 bootloader 0x1000 C:/Users/roger/source/repos/stb-r1/build/bootloader/bootloader.bin"
Bootloader binary size 0x7ec0 bytes. 0x140 bytes (1%) free.
[3/4] Generating C:/Users/roger/esp4.4B/esp-idf/components/esp_littlefs/mklittlefs/mklittlefs
make: Nothing to be done for 'dist'.
[4/4] cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\esp-idf\main && C:\Users\roger\...b-r1/spiffsImage -s 0xf0000 -p 256 -b 4096 C:/Users/roger/source/repos/stb-r1/build/storage.bin" 
FAILED: esp-idf/main/CMakeFiles/littlefs_storage_bin
cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\esp-idf\main && C:\Users\roger\esp4.4B\esp-idf\components\esp_littlefs\mklittlefs\mklittlefs -d 0 -c C:/Users/roger/source/repos/stb-r1/spiffsImage -s 0xf0000 -p 256 -b 4096 C:/Users/roger/source/repos/stb-r1/build/storage.bin"
ninja: build stopped: subcommand failed.`

BrianPugh, 2022-07-14:
Hi @jeeper1974 ! Odd to see it failing; is there any additional information outputted?

jeeper1974, 2022-07-15:
That is all it outputs. I've got a folder named "spiffsImage" that is one directory above the "main" project so I set inCMakeLists.txt: 

`littlefs_create_partition_image(storage` ../spiffsImage)`

My partition table looks like this:
```
nvs,data,nvs,,0x6000
phy_init,data,phy,,0x1000
factory,0,0,,2M
storage,data,spiffs,,1M
```

But when I compile it only fails with:
```
[1/8] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\bootloader\esp-idf\esptool_py && C:\Users\roger\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/roger/esp4.4B/esp-idf/components/partition_table/check_sizes.py --offset 0x9000 bootloader 0x1000 C:/Users/roger/source/repos/stb-r1/build/bootloader/bootloader.bin"
Bootloader binary size 0x7ec0 bytes. 0x140 bytes (1%) free.
[2/6] Generating C:/Users/roger/esp4.4B/esp-idf/components/esp_littlefs/mklittlefs/mklittlefs
FIND: Parameter format not correct
make: Nothing to be done for 'dist'.
[3/6] cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\esp-idf\main && C:\Users\roger\esp4.4B\esp-idf\comp...ger/source/repos/stb-r1/spiffsImage -s 0x100000 -p 256 -b 4096 C:/Users/roger/source/repos/stb-r1/build/storage.bin"
FAILED: esp-idf/main/CMakeFiles/littlefs_storage_bin
cmd.exe /C "cd /D C:\Users\roger\source\repos\stb-r1\build\esp-idf\main && C:\Users\roger\esp4.4B\esp-idf\components\esp_littlefs\mklittlefs\mklittlefs -d 0 -c C:/Users/roger/source/repos/stb-r1/spiffsImage -s 0x100000 -p 256 -b 4096 C:/Users/roger/source/repos/stb-r1/build/storage.bin"
[4/6] Linking CXX executable stb_fw.elf
ninja: build stopped: subcommand failed.
```
I ended up just switching back to SPIFFS which is working fine.  If you have any suggestions I can try out, let me know and I'll give littlefs a go again. Thanks

BrianPugh, 2022-07-18:
I'm willing to bet [this line](https://github.com/BrianPugh/mklittlefs/blob/a14dabe444e9a265aab04d085d8ca8f8536d799b/Makefile#L61) is causing the issue, but not quite sure what actions to take. I can maybe try investigating this weekend, but I might be a bit slow to get around to this.

Mr-Techtron, 2023-02-18:
any update on this? I am getting the same error on my Windows 11, using eclipse as an IDE.

jeeper1974, 2023-02-19:
Sorry. I switched back to SPIFF and have not tried again. 

 

Roger

Mr-Techtron, 2023-02-19:
@jeeper1974 
Hello jeeper, just wanted to let you know that I managed to compile the project in windows 11. I followed the workaround mentioned in #102 . For gcc and make, I used MSYS2.