joltwallet/esp_littlefs #86: Failed to build littlefs with esp-idf-v5.0-dev

AxelLin, 2022-07-24:
After pulling the latest esp-idf master update, I got below build error:
In file included from /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/src/esp_littlefs.c:16:
/home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/src/littlefs_api.h:9:10: fatal error: esp_vfs.h: No such file or directory
9 | #include "esp_vfs.h"
| ^~~~~~~~~~~
compilation terminated.

I have reported this to https://github.com/espressif/esp-idf/issues/9421, the littlefs needs fix.

BrianPugh, 2022-07-25:
I'll need to update CI for esp-idf v5.0 to make sure stuff doesn't break in the future.

closed 2022-07-25 by BrianPugh

BrianPugh, 2022-07-29:
added more ci in https://github.com/joltwallet/esp_littlefs/pull/88

Need to fix up some stuff so it builds on latest