joltwallet/esp_littlefs #96: Suggestion: Add version in source code too.

7aman, 2022-08-11:
It is common that the version of library some how included in the source code.  
It could be a comment in one or all source and headers, like something FreeRTOS does in header of all files:  
```C
/*
 * FreeRTOS Kernel V10.4.1
 * Copyright (C) 2020 Amazon.com, Inc. or its affiliates.  All Rights Reserved.
 * [continues with copyright and license ]
 */
```
Or it could be a macro, like LittleFS itself or FreeRTOS:  
```C
/* in task.h */
#define tskKERNEL_VERSION_NUMBER       "V10.4.1"
#define tskKERNEL_VERSION_MAJOR        10
#define tskKERNEL_VERSION_MINOR        4
#define tskKERNEL_VERSION_BUILD        1
```

This way it is always clear that which version of this library is added to a project. It is especially helpful for the ones that do not use component manager.

A side note: in `README.md` in `How tot use` section, it still points to `v1.3.1`. I was not sure it stands as an example there or it missed its increasing during version bumps.

BrianPugh, 2022-08-11:
I'll add this! I just need to figure out how to setup CI to do all of this so that its not up to me remembering. If you know of any other projects that has CI do this, it'll make my job easier 😄 .

BrianPugh, 2022-08-12:
Addressed in v1.5.0

* `ESP_LITTLEFS_VERSION_NUMBER` - string "MAJOR.MINOR.PATCH"
* `ESP_LITTLEFS_VERSION_MAJOR` - integer major value
* `ESP_LITTLEFS_VERSION_MINOR` - integer minor value
* `ESP_LITTLEFS_VERSION_PATCH` - integer patch value

closed 2022-08-12 by BrianPugh