joltwallet/esp_littlefs #97: Crash when reading a file opened with unsupported POSIX flags

arex-ebee, 2022-08-11:
While migrating to LittleFS within an existing project I observed crashes caused by assertions checking for `LFS_O_RDONLY` file flag set within LittleFS code. Reason for this seems to be that the conversion from POSIX flags to LittleFS flags within the appropriate `open()` VFS call fails when a file is opened with flags that are not relevant for LittleFS.

Example:
```
int fd = open("/lfs_mountpoint/filename.txt", O_RDONLY | O_NONBLOCK | O_CLOEXEC);
read(fd, buf, 16);
```
This will end up in `esp_littlefs.c:vfs_littlefs_open()` with the given flags. The function calls `esp_littlefs_flags_conv()` converting the POSIX flags to LittleFS flags. But as there is not only `O_RDONLY` but also additional flags that are not handled, the resulting LittleFS flag is 0. Later on LittleFS checks for `LFS_O_RDONLY` when calling `lfs_file_rawread()` which is not set in that case.

I would consider this as a weakness inside LittleFS not checking for correct flags during open call. But the faulty conversion is definitely a bug in esp_littlefs.

I solved it by masking out all unsupported flags in `esp_littlefs_flags_conv()` prior to handling them in:

```
static int esp_littlefs_flags_conv(int m) {
    int lfs_flags = 0;

    // Mask out unsupported flags as otherwise they could lead to errors with below code
    m &= (O_APPEND | O_RDONLY | O_WRONLY | O_RDWR | O_EXCL | O_CREAT | O_TRUNC);;

    if (m == O_APPEND) {ESP_LOGV(TAG, "O_APPEND"); lfs_flags |= LFS_O_APPEND;}
    if (m == O_RDONLY) {ESP_LOGV(TAG, "O_RDONLY"); lfs_flags |= LFS_O_RDONLY;}
    if (m & O_WRONLY)  {ESP_LOGV(TAG, "O_WRONLY"); lfs_flags |= LFS_O_WRONLY;}
    if (m & O_RDWR)    {ESP_LOGV(TAG, "O_RDWR");   lfs_flags |= LFS_O_RDWR;}
    if (m & O_EXCL)    {ESP_LOGV(TAG, "O_EXCL");   lfs_flags |= LFS_O_EXCL;}
    if (m & O_CREAT)   {ESP_LOGV(TAG, "O_CREAT");  lfs_flags |= LFS_O_CREAT;}
    if (m & O_TRUNC)   {ESP_LOGV(TAG, "O_TRUNC");  lfs_flags |= LFS_O_TRUNC;}
    return lfs_flags;
}
```

BrianPugh, 2022-08-11:
seems reasonable, I'll make a PR soon

BrianPugh, 2022-08-12:
PR merged; will release soon.

closed 2022-08-12 by BrianPugh

BrianPugh, 2022-08-12:
released in v1.5.0