joltwallet/esp_littlefs #102: Building on windows 11

eldendiss, 2022-08-27:
Hello,  
I stumbled upon issues with mklittlefs when trying to build a esp-idf project on Windows 11. I decided to post it here, so my fellow corporate rats stuck with windows can save yourself some frustration.

1st issue:
```
$ make dist
FIND: Parameter format not correct
cc -std=gnu99 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -I ../src -D VERSION=\"0.2.3-40-ga14dabe\" -D LITTLEFS_VERSION=\"v2.5.0\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o ../src/littlefs/lfs.o ../src/littlefs/lfs.c
process_begin: CreateProcess(NULL, cc -std=gnu99 -Os -Wall -Itclap -Iinclude -Ilittlefs -I. -I ../src -D VERSION=\"0.2.3-40-ga14dabe\" -D LITTLEFS_VERSION=\"v2.5.0\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32 -c -o ../src/littlefs/lfs.o ../src/littlefs/lfs.c, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [../src/littlefs/lfs.o] Error 2
```

Simple fix, make tries to use cc as a compiler, but it is not present in mingw. So you need to create a symlink:
```
mklink c:\MinGW\bin\cc c:\MinGW\bin\gcc
```

Then you will find yourself in front of another error:

```
$ make dist
FIND: Parameter format not correct
g++ main.o ../src/littlefs/lfs.o ../src/littlefs/lfs_util.o -o mklittlefs  
strip mklittlefs
strip: 'mklittlefs': No such file
make: *** [mklittlefs] Error 1
```

I noticed that there is .exe extension missing. So I took a glance at mklittlefs makefile, and i found out that TARGET_OS variable is equal to win32, but we are checking if it is equal to windows.  
So in Makefile at line 29, you need to change this:

```
ifeq ($(TARGET_OS),windows)
	ARCHIVE ?= zip
	TARGET := mklittlefs.exe
	TARGET_CFLAGS = -mno-ms-bitfields
	TARGET_LDFLAGS = -Wl,-static -static-libgcc -static-libstdc++
else
	ARCHIVE ?= tar
	TARGET := mklittlefs
endif
````

To this:

```
ifeq ($(TARGET_OS),win32)
	ARCHIVE ?= zip
	TARGET := mklittlefs.exe
	TARGET_CFLAGS = -mno-ms-bitfields
	TARGET_LDFLAGS = -Wl,-static -static-libgcc -static-libstdc++
else
	ARCHIVE ?= tar
	TARGET := mklittlefs
endif
```

And finally just one error:
```
make dist
FIND: Parameter format not correct
cp mklittlefs.exe mklittlefs-0.2.3-40-ga14dabe-generic-win32/
zip -r mklittlefs-0.2.3-40-ga14dabe-generic-win32.zip mklittlefs-0.2.3-40-ga14dabe-generic-win32
process_begin: CreateProcess(NULL, zip -r mklittlefs-0.2.3-40-ga14dabe-generic-win32.zip mklittlefs-0.2.3-40-ga14dabe-generic-win32, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [mklittlefs-0.2.3-40-ga14dabe-generic-win32.zip] Error 2
```

This means that the binary zip isnt present. Luckily, windows 11 natively supports tar, so you just need to change line 30 to:

```
ARCHIVE ?= tar
```

There is still an issue with `FIND: Parameter format not correct` being present in stdout, but it seems that the binaries are compiled sucessfuly nevertheless.

**TLDR Set symlink cc<===>gcc and use modified Makefile working on Windows 11:**
```
# OS detection. Not used in CI builds
ifndef TARGET_OS
ifeq ($(OS),Windows_NT)
	TARGET_OS := win32
else
	UNAME_S := $(shell uname -s)
	ifeq ($(UNAME_S),Linux)
		UNAME_M := $(shell uname -m)
		ifeq ($(UNAME_M),x86_64)
			TARGET_OS := linux64
		endif
		ifeq ($(UNAME_M),i686)
			TARGET_OS := linux32
		endif
		ifeq ($(UNAME_M),armv6l)
			TARGET_OS := linux-armhf
		endif
	endif
	ifeq ($(UNAME_S),Darwin)
		TARGET_OS := osx
	endif
	ifeq ($(UNAME_S),FreeBSD)
		TARGET_OS := freebsd
	endif
endif
endif # TARGET_OS

# OS-specific settings and build flags
ifeq ($(TARGET_OS),win32)
	ARCHIVE ?= tar
	TARGET := mklittlefs.exe
	TARGET_CFLAGS = -mno-ms-bitfields
	TARGET_LDFLAGS = -Wl,-static -static-libgcc -static-libstdc++
else
	ARCHIVE ?= tar
	TARGET := mklittlefs
endif

# Packaging into archive (for 'dist' target)
ifeq ($(ARCHIVE), zip)
	ARCHIVE_CMD := zip -r
	ARCHIVE_EXTENSION := zip
endif
ifeq ($(ARCHIVE), tar)
	ARCHIVE_CMD := tar czf
	ARCHIVE_EXTENSION := tar.gz
endif

STRIP ?= strip

VERSION ?= $(shell git describe --always)
LITTLEFS_VERSION := $(shell git -C ../src/littlefs describe --tags || echo "unknown")
BUILD_CONFIG_NAME ?= -generic

OBJ		:= main.o \
		   ../src/littlefs/lfs.o \
		   ../src/littlefs/lfs_util.o

INCLUDES := -Itclap -Iinclude -Ilittlefs -I. -I ../src

FILES_TO_FORMAT := $(shell find . -not -path './littlefs/*' \( -name '*.c' -o -name '*.cpp' \))

DIFF_FILES := $(addsuffix .diff,$(FILES_TO_FORMAT))

# clang doesn't seem to handle -D "ARG=\"foo bar\"" correctly, so replace spaces with \x20:
BUILD_CONFIG_STR := $(shell echo $(CPPFLAGS) | sed 's- -\\\\x20-g')

override CPPFLAGS := \
	$(INCLUDES) \
	-D VERSION=\"$(VERSION)\" \
	-D LITTLEFS_VERSION=\"$(LITTLEFS_VERSION)\" \
	-D BUILD_CONFIG=\"$(BUILD_CONFIG_STR)\" \
	-D BUILD_CONFIG_NAME=\"$(BUILD_CONFIG_NAME)\" \
	-D __NO_INLINE__ \
	-D LFS_NAME_MAX=32 \
	$(CPPFLAGS)

override CFLAGS := -std=gnu99 -Os -Wall $(TARGET_CFLAGS) $(CFLAGS)
override CXXFLAGS := -std=gnu++11 -Os -Wall $(TARGET_CXXFLAGS) $(CXXFLAGS)
override LDFLAGS := $(TARGET_LDFLAGS) $(LDFLAGS)

DIST_NAME := mklittlefs-$(VERSION)$(BUILD_CONFIG_NAME)-$(TARGET_OS)
DIST_DIR := $(DIST_NAME)
DIST_ARCHIVE := $(DIST_NAME).$(ARCHIVE_EXTENSION)

all: $(TARGET)

dist: $(DIST_ARCHIVE)

$(DIST_ARCHIVE): $(TARGET) $(DIST_DIR)
	cp $(TARGET) $(DIST_DIR)/
	$(ARCHIVE_CMD) $(DIST_ARCHIVE) $(DIST_DIR)

$(TARGET): $(OBJ)
	$(CXX) $^ -o $@ $(LDFLAGS)
	$(STRIP) $(TARGET)

$(DIST_DIR):
	@mkdir -p $@

clean:
	@rm -f $(TARGET) $(OBJ) $(DIFF_FILES)

format-check: $(DIFF_FILES)
	@rm -f $(DIFF_FILES)

.PHONY: all clean dist format-check
```

BrianPugh, 2022-08-28:
hi @eldendiss thanks for all the findings! I'll try and work on this soon. As you've noticed, `mklittlefs ` certainly needs some attention.

BrianPugh, 2023-05-03:
Implementing these changes now; reading other Issues to see if they all amount to the same thing.

PR #123 (2023-05-03): Windows-compatible mklittlefs
closes #102
closed 2023-05-03 by BrianPugh