joltwallet/esp_littlefs #106: Project fails to compile when  LITTLEFS included in an ESP-IDF project, from V1.5.1 as a component.

wreyford, 2022-11-17:
It took met the best part of 6 hours to figure this one out.
I narrowed it down to the LittleFS component.

I'm getting the following compiler error, when building an project:

`Python requirements are satisfied.
Processing 7 dependencies:
[1/7] espressif/esp_lcd_touch (1.0.2)
[2/7] espressif/esp_lcd_touch_ft5x06 (1.0.2)
[3/7] espressif/esp_websocket_client (0.0.4)
[4/7] espressif/led_strip (2.0.0)
[5/7] idf (5.1.0)
[6/7] joltwallet/littlefs (1.5.1)
[7/7] lvgl/lvgl (8.3.0)
CMake Error at C:/Espressif/frameworks/esp-idf-master/tools/cmake/build.cmake:263 (message):
-- Configuring incomplete, errors occurred!
See also "C:/Espressif/workspace/LoadAssist-S3/build/CMakeFiles/CMakeOutput.log".
  Failed to resolve component 'esp_partition'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-master/tools/cmake/build.cmake:305 (__build_resolve_and_add_req)
  C:/Espressif/frameworks/esp-idf-master/tools/cmake/build.cmake:306 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-master/tools/cmake/build.cmake:592 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-master/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:8 (project)

cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.`

At first I didn't know what was causing the error.
So started a new project from ESP-IDF examples, and I chose the STATION example.
I added the LittleFS component to the project using Espressif IDE V2.7.0, esp-idf master.
If I then build, the station project, I also get this error.
This way, I could at least isolate the component as the problem, possibly due to breaking changes in the esp-idf master.

esp_partition.h resides in ..\components\spi_flash\include\esp_partition.h

I tried adding spi_flash to my CMakeList.txt,
but this didn't help either, i.e. REQUIRES spi_flash
I also tried as a PRIV_REQUIRES, but no luck.

Please could you assist with this problem, as my entire project relies very heavily on LittleFS.

Thx

wreyford, 2022-11-17:
I notcie that esp_partition in included in littlefs_api.h

BrianPugh, 2022-11-17:
I reran the CI workflow, stuff still works there:

https://github.com/joltwallet/esp_littlefs/actions/runs/3449046910

wreyford, 2022-11-19:
Perhaps it is specific to the way the Espressif -IDE compiler deals with the headers.
I just cannot get tit to compile.. I've tried everything!
Please try it out if you have some time, in the IDE, open an example Station, then just right click on the project and add the LittleFS dependency, and compile. You will get the compiler error.

[image]

wreyford, 2022-11-19:
Your Cmake does include it as a PRIV_REQUIRES through pr.

[image]

BrianPugh, 2022-11-19:
oh, i think i know what this is; you probably need to update your esp-idf; as `esp_partition` [was recently made into its own independent component.](https://github.com/espressif/esp-idf/commit/c9c7573f715534edf19e8290bc10d2a5d134c4a8)

Growing pains of being on the bleeding edge :P

wreyford, 2022-11-19:
Thx, I actually came to the same conclusion, I have been updating ESP-IDF with the built in menu option in Espressif IDE and turns out it is for some reason not doing what it is supposed to. So ended up re-installing master manually. I see they already moved it 17 days ago, (and I have done many updates over this period) moved esp_partition to its own component., and I quite happily was updating using tbe IDE's menu and didn 't suspect anything. Sorry to have waisted your time. ThxSent from my phone-------- Original message --------From: Brian Pugh ***@***.***>Date: Sat, 19 Nov 2022, 19:21To: joltwallet/esp_littlefs ***@***.***>Cc: wreyford ***@***.***>, Author ***@***.***>Subject: Re: [joltwallet/esp_littlefs] Project fails to compile when  LITTLEFS included in an ESP-IDF project, from V1.5.1 as a component. (Issue #106)
oh, i think i know what this is; you probably need to update your esp-idf; as esp_partition was recently made into its own independent component.
Growing pains of being on the bleeding edge :P

BrianPugh, 2022-11-19:
no worries, glad you were able to resolve your issue!

closed 2022-11-19 by BrianPugh