joltwallet/esp_littlefs #108: Increasing read and write speeds on LittleFS ESP-IDF (ESP32-S3)

Nickobellic15X, 2022-12-20:
Hello,

I was wondering how I could increase the read and more importantly the write speeds on littlefs. My application writes large files (images 5MP) to the flash to store them for some time before sending it to a backend server via https. 

Currently it takes about 3.5 to 4 seconds to write an image roughly 400kB in size. Is there any way I can increase this speed?

Thank you.

BrianPugh, 2022-12-20:
In your project configuration (this is relevant to the esp32, i've never used the esp32-s3), make sure your flash is set to (if it supports it) 80MHz (double the default 40MHz) and is set to QIO mode. I believe doing both should make your flash access roughly 4x faster.

To get the absolute fastest speed, you might just want to directly write to a partition without a filesystem (i.e. without `esp_littlefs`). If you are only caching a single image at a time, this should be pretty easy. I'm not sure how much overhead littlefs has on the writing, but it'd be a good comparison. I imagine that compared to the hardware configurations above, this improvement will be relatively minimal (I'd be surprised if it's more than 5% faster).

Nickobellic15X, 2022-12-21:
Right, will take note of it. The current challenge is to see what configurations of [SPI flash and PSRAM work](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#flash-psram-combination).

After which I will run a few image capture cycles and time it. I'll post my findings here later.

p3zhy, 2023-02-05:
Hello,
I want to use QIO mode in esp32s3 (Winbond 25Q64 SpiFlash).
This is my sdkconfig  : 
`
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_FLASHMODE_QIO=y
`
I can't init littlefs (esp_vfs_littlefs_register).
 
I am facing these errors : 

IDF/components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
W (1174) esp_littlefs: mount failed,  (-84). formatting...
IDF/components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x0
IDF/components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x0 has become unwritable
E (19624) esp_littlefs: Failed to format filesystem
E (19634) esp_littlefs: format failed
E (19634) esp_littlefs: Failed to initialize LittleFS