joltwallet/esp_littlefs #109: Suggestion: adding Kconfig options for LittleFS debug level.

everslick, 2023-01-11:
As documented here: https://github.com/espressif/esp32-arduino-lib-builder/issues/103 and here: https://github.com/espressif/arduino-esp32/issues/7700 it would be nice, if we could disable serial output in a clean and consistent way.

igrr, 2023-01-11:
And as a follow-up, an alternative suggestion is to define littleFS logging macros to ESP-IDF ones. Then the log level can be directly controlled via ESP-IDF configuration options, esp_log_level_set, and log redirection will also work.

BrianPugh, 2023-01-12:
thanks guys! I'm currently pretty busy, but totally open to pull requests. Otherwise, I'll try and get to this sometime in the next few weeks.

BrianPugh, 2023-03-01:
@frozen-eye Not sure what your schedule is like, but this might be relevant to your last PR. I haven't had the time to work on this, but would gladly accept PRs.

BrianPugh, 2023-05-03:
Will be included in upcoming 1.5.4

PR #122 (2023-05-03): unify logging tags; use esp-loggers in lfs_config instead of printf
Should resolve #109
closed 2023-05-03 by BrianPugh