joltwallet/esp_littlefs #110: Support for ESP8266

colesnicov, 2023-01-25:
hello. One question please. will be worked under esp8266-rtos-sdk. Thanx.

BrianPugh, 2023-01-25:
not sure, I've only tested it under esp-idf.

closed 2023-01-25 by colesnicov

colesnicov, 2023-01-25:
I tried to compile but without success:

```
$ make 
Toolchain path: /home/denis/esp/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc
Toolchain version: esp-2020r3-49-gd5524c1
Compiler version: 8.4.0
Python requirements from /home/denis/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
Project is not inside a git repository, or git repository has no commits
will not use 'git describe' to determine PROJECT_VER.
App "ota" version: 1
CC build/esp_littlefs/src/esp_littlefs.o
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c: In function 'esp_littlefs_init':
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:575:16: error: unknown type name 'portMUX_TYPE'
         static portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
                ^~~~~~~~~~~~
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:575:35: error: 'portMUX_INITIALIZER_UNLOCKED' undeclared (first use in this function)
         static portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:575:35: note: each undeclared identifier is reported only once for each function it appears in
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:576:32: error: macro "portENTER_CRITICAL" passed 1 arguments, but takes just 0
         portENTER_CRITICAL(&mux);
                                ^
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:576:9: error: 'portENTER_CRITICAL' undeclared (first use in this function)
         portENTER_CRITICAL(&mux);
         ^~~~~~~~~~~~~~~~~~
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:581:31: error: macro "portEXIT_CRITICAL" passed 1 arguments, but takes just 0
         portEXIT_CRITICAL(&mux);
                               ^
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:581:9: error: 'portEXIT_CRITICAL' undeclared (first use in this function)
         portEXIT_CRITICAL(&mux);
         ^~~~~~~~~~~~~~~~~
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:575:29: warning: unused variable 'mux' [-Wunused-variable]
         static portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;
                             ^~~
At top level:
/home/denis/projects/ota/components/esp_littlefs/src/esp_littlefs.c:575:29: warning: 'mux' defined but not used [-Wunused-variable]
make[1]: *** [/home/denis/esp/ESP8266_RTOS_SDK/make/component_wrapper.mk:292: src/esp_littlefs.o] Chyba 1
make: *** [/home/denis/esp/ESP8266_RTOS_SDK/make/project.mk:571: component-esp_littlefs-build] Chyba 2
```