joltwallet/esp_littlefs #114: esp_littlefs using external flash (VSPI)

TravisRo, 2023-03-02:
I have this working with an external 16MB flash IC connected to the VSPI bus.  To get it to work I had to register the partition using the FAT subtype instead of SPIFFS.

EG:
```c
static const esp_partition_t* fs_add_partition(esp_flash_t* ext_flash, const char* partition_label)
{
	ESP_LOGI(FS_TAG, "Adding external Flash as a partition, label=\"%s\", size=%d KB", partition_label, ext_flash->size / 1024);
	const esp_partition_t* partition;
	ESP_ERROR_CHECK(esp_partition_register_external(ext_flash, 0, ext_flash->size, partition_label, ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_FAT, &partition));
	return partition;
}
```

Are there any side affects of doing it this way?  I'm not a filesystem guru so I don't know what kind of ramifications this has or exactly what the differences are between using `ESP_PARTITION_SUBTYPE_DATA_FAT` and `ESP_PARTITION_SUBTYPE_DATA_SPIFFS`

Thanks,
Travis

BrianPugh, 2023-03-02:
I'm not sure it has any impact; I *think* it might impact other functions like `esp_partition_find`. I'm actually surprised it didn't work with `ESP_PARTITION_SUBTYPE_DATA_SPIFFS `. If you find any relevant information, I'd be curious to know.

TravisRo, 2023-03-02:
My bad, that doesn't seem to be what ails it.  It happens when I switch from the standard esp FAT file system that has a few files to littlefs.  That watchdog triggers several reboots then it eventually succeeds at what ever it is trying to do (format?)

```
E (25160) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40085ED1:0x3FFB1190 0x40082A1D:0x3FFB11B0 0x4000BFED:0x3FFCD480 0x4008AE1A:0x3FFCD490 0x400E4E10:0x3FFCD4B0 0x400E4E1F:0x3FFCD4D0 0x400D242E:0x3FFCD4F0 0x40088E6C:0x3FFCD510

E (30160) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (30160) task_wdt:  - IDLE (CPU 0)
E (30160) task_wdt: Tasks currently running:
E (30160) task_wdt: CPU 0: main
E (30160) task_wdt: CPU 1: IDLE
E (30160) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E4B63:0x3FFB0B80 0x40082A1D:0x3FFB0BB0 0x40008544:0x3FFCC320 0x40085D49:0x3FFCC340 0x4008DB05:0x3FFCC360 0x4008E06D:0x3FFCC390 0x4008572B:0x3FFCC3D0 0x400E446A:0x3FFCC420 0x400D7AF2:0x3FFCC440 0x400D7D78:0x3FFCC470 0x400D7EC3:0x3FFCC4B0 0x400D56DA:0x3FFCC580 0x400F4112:0x3FFCC660
```

If I uncheck "Watch CPU 0 Idle Task" and "Watch CPU 1 Idle Task" in menuconfig then it works but it takes approx 30s for whatever it is doing to finish. Does it attempt some kind of recovery when it thinks the file system exists but has been fouled up?

Let me know if there is anything else I can do to help.  We are currently using the esp fat filesystem in a product and the word "fragile" does sum it up quite nicely. :)

I am using espidf 4.4.4 with a winbond 16MB QIO flash IC on a 2nd generation (dual core) ESP32-MINI-01 devkit 

I'll do some more testing in the AM.

Thanks,
Travis

BrianPugh, 2023-03-02:
When switching file systems, it'll perform a format. We'll probably have to kick the watchdog somewhere during the formatting process; it hasn't been an issue with the standard flash on most esp32's, so its probably only now becoming an issue with the large second flash.

BrianPugh, 2023-03-02:
To give more detail to my previous comment:

I'm assuming that [it's one of these two lines](https://github.com/joltwallet/esp_littlefs/blob/fb1470a59dad5949396919f34b9856fa8d29d55f/src/esp_littlefs.c#L377) is taking a long time and is causing a timeout.

```
esp_littlefs_erase_partition(partition_label);  // Line 1 (I think 80% chance it's this)
res = lfs_format(efs->fs, &efs->cfg);           // Line 2 (I think 20% chance it's this)
```

## If it's line 1
I think you are running into the same issue here:
https://github.com/espressif/esp-idf/issues/9212

Which seems like it has been fixed on Jan 12, 2023:
https://github.com/espressif/esp-idf/commit/cfc833283320c1f84de00274d0754c77232d369c

So if you are using the latest of esp-idf, that might work.
We use `esp_partition_erase_range`, which internally invokes the discussed`spi_flash_erase_range` (unless `CONFIG_SPI_FLASH_USE_LEGACY_IMPL=y`).

There *may* be workarounds if you cannot update esp-idf, but I think this starts to get outside the scope of lfs.

## If it's line 2
Then we'll need to feed the watchdog or yield

```
// WDT Solution 1
#include "esp_task_wdt.h"

esp_task_wdt_reset();  // We might have to shove this call into lfs code, which is unfortunate.
```

```
// WDT Solution 2
vTaskDelay(1);  // Call this every so often.
```

Overall, I *think* WDT solution 2 is preferred (plays nicer with other tasks), but it would still involve us putting custom code into the LFS source.

TravisRo, 2023-03-02:
Thanks a lot for the extra info!  I can't use espidf 5.x yet but I'll poke around in the lfs source and see if I can come up with an elegant solution.  If I find one, I'll submit a pull request with the fix.

Using the standard esp FAT filesystem, the format operation is extremely fast (less than 1s) which makes me think it's not doing much.

BrianPugh, 2023-03-02:
Actually, my `esp_littlefs_erase_partition` call might be totally redundant and unnecessary. As I think `lfs_format` will do the necessary actions. So maybe try commenting that out? Erasing the whole partition (if unnecessary) is so slow. If its unnecessary, the first-time-mount performance boost would be a win for all users!

TravisRo, 2023-03-02:
I'll give it a try but when I look in the code, it ultimately ends up calling "esp_flash_erase_region" which calls a yield function before each block/sector erase. 
IE:
```C
//check before the operation, in case this is called too close to the last operation
if (chip->chip_drv->yield) {
    err = chip->chip_drv->yield(chip, 0);
    if (err != ESP_OK) {
        return err;
    }
}

```

TravisRo, 2023-03-03:
It is in-fact the `esp_littlefs_erase_partition` function which simply calls a espidf function. I can reproduce the problem simply by calling:
```C
esp_partition_erase_range(fs_ext_partition, 0, fs_ext_partition->size);
```

..so.. not really a littlefs problem at all

BrianPugh, 2023-03-03:
well thats good to hear on my end :D. Did you try updating esp-idf? Because it may have been fixed relatively recently (in commit from previous comment).

TravisRo, 2023-03-03:
I'm don't think that patch got pulled into the 4.4 branch.  It looks like it just went into "master" but the problem seems to be that the "yield" function that it's calling is `spi_flash_chip_generic_yield` which doesn't do anything because `chip->os_func->check_yield` is null!  If your running on SPI1 it uses a different set of "yield" functions.

```C
esp_err_t spi_flash_chip_generic_yield(esp_flash_t* chip, uint32_t wip)
{
    esp_err_t err = ESP_OK;
    uint32_t flags = wip? 1: 0; //check_yield() and yield() impls should not issue suspend/resume if this flag is zero

    if (chip->os_func->check_yield) {
        uint32_t request;
        //According to the implementation, the check_yield() function may block, poll, delay or do nothing but return
        err = chip->os_func->check_yield(chip->os_func_data, flags, &request);
        if (err == ESP_OK) {
            if (err == ESP_OK && (request & SPI_FLASH_YIELD_REQ_YIELD) != 0) {
                uint32_t status;
                //According to the implementation, the yield() function may block until something happen
                err = chip->os_func->yield(chip->os_func_data, &status);
            }
        } else if (err == ESP_ERR_TIMEOUT) {
            err = ESP_OK;
        } else {
            abort();
        }
    }
    return err;
}
```

I will do some testing without the call to `esp_littlefs_erase_partition ` as you suggested.

TravisRo, 2023-03-03:
It does not appear to be necessary to to call `esp_littlefs_erase_partition` and that will DRASTICALLY improve your first-time initialize time as well as eliminate the problem I was having which imo, is a bug in the espidf framework.  (they assign a `yield` function but not a `check_yield` function)

Perhaps you'll want to do more to verify that `esp_littlefs_erase_partition` really isn't needed.  I started with a esp FAT file system that had 16 1,000,000 byte files each filled with completely random data.  I then swap out the esp fat firmware with a variation of the littlefs demo app and it worked flawlessly.

The littlefs format function ends up calling the yield function (which does nothing in my case) 2 times which is not nearly enough to trigger the wdt in any case.

It took  96834 uS to format a 16MB partition using QIO running at only 20mhz.

Thanks for the great port!  I'm hoping it will fix our corruption issues caused by power loss.

Regards,
Travis

```
I (20159) FS: Initializing external SPI Flash
I (20159) FS: Pin assignments:
I (20159) FS: MOSI: 23   MISO: 19   SCLK: 18   CS:  5
I (20159) spi_flash: detected chip: winbond
I (20159) spi_flash: flash io: qio
I (20159) FS: Initialized external Flash, size=16384 KB, ID=0xef4018
I (20159) FS: Adding external Flash as a partition, label="littlefs", size=16384 KB
I (20164) FS: Listing data partitions:
I (20164) FS: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB
I (20164) FS: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB
I (20164) FS: - partition 'littlefs', subtype 130, offset 0x0, size 16384 kB
I (20169) FS: esp_vfs_littlefs_register start
./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
W (20169) esp_littlefs: mount failed,  (-84). formatting...
I (20174) spi_flash: YIELD!
I (20219) spi_flash: YIELD!
I (20264) FS: esp_vfs_littlefs_register end
I (20264) FS: esp_vfs_littlefs_register took 96834 us
I (20264) FS: littlefs registered successfully!
I (20264) Main: Partition size: total: 16777216, used: 8192
I (20269) Main: Opening file
I (20269) Main: File written
I (20269) Main: Renaming file
I (20274) Main: Reading file
I (20274) Main: Read from file: 'LittleFS Rocks!'
```

BrianPugh, 2023-03-03:
I'll remove the `esp_littlefs_erase_partition`, logically there's no real reason for it. You could make an argument that it's to securely delete stuff; but that's really out of the scope for this project, so no need for it.

BrianPugh, 2023-03-03:
I opened up PR #115 that removes this partition erasing. I'm not going to merge it yet because I need to run the unit test suite, and I currently don't have my esp hardware handy.

BrianPugh, 2023-05-03:
merged and will be part of v1.5.4 release.

closed 2023-05-03 by BrianPugh