joltwallet/esp_littlefs #116: Panic on constant file write

tsctrl, 2023-03-08:
Hi,

Thanks for this great library and work well. I am having some issue that i cant figure out why.

1. I have a que task that constantly feed logs and writing a file.log file using arduino File.write(buff) function;
2. I have other task that manually running and reading json files, File.open(path, "r")

The No 1 task keep causing me a panic with error as below:
```
==================== THREAD 1 (TCB: 0x3ffc8750, name: 'readLogQueue') =====================
#0  0x40081a4d in panic_abort (details=0x3ffc83a4 \"assert failed: lfs_file_rawwrite lfs.c:3415 ((file->flags & LFS_O_WRONLY) == LFS_O_WRONLY)\") at esp_system/panic.c:402
#1  0x40088e8c in esp_system_abort (details=0x3ffc83a4 \"assert failed: lfs_file_rawwrite lfs.c:3415 ((file->flags & LFS_O_WRONLY) == LFS_O_WRONLY)\") at esp_system/esp_system.c:128
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```
```
==================== THREAD 3 (TCB: 0x3ffe76f8, name: 'loadConfigJson') =====================
#0  0x400856b0 in esp_crosscore_int_send_yield (core_id=0) at esp_system/crosscore_int.c:144
#1  0x40089f18 in xQueueSemaphoreTake (xQueue=0x3ffc5ccc, xTicksToWait=<optimized out>) at hal/esp32/include/hal/cpu_ll.h:39
#2  0x401df14c in xQueueTakeMutexRecursive (xMutex=0x3ffc5ccc, xTicksToWait=4294967295) atfreertos/queue.c:724
#3  0x40129741 in vfs_littlefs_pread (ctx=<optimized out>, fd=<optimized out>, dst=<optimized out>, size=<optimized out>, offset=<optimized out>) at esp_littlefs/src/esp_littlefs.c:1206
#4  0x400d4ce1 in esp_vfs_read (r=<optimized out>, fd=<optimized out>, dst=0x3ffc74ac, size=128) at vfs/vfs.c:457
#5  0x4000bdbe in ?? ()
#6  0x40001128 in ?? ()
#10 0x40124960 in VFSFileImpl::read (this=<optimized out>, buf=0x3fff174f \":,N\\021@\\a\", size=1) at arduino/libraries/FS/src/vfs_api.cpp:383
#11 0x40114e20 in fs::File::read (this=0x40124930 <VFSFileImpl::write(unsigned char const*, unsigned int)+52>, buf=0x3fff174f \":,N\\021@\\a\", size=1) at arduino/libraries/FS/src/FS.cpp:78
#12 0x40114e38 in fs::File::read (this=<optimized out>, buf=0x3fff174f \":,N\\021@\\a\", size=1) at xtensa-esp32-elf\\include\\c++\\8.4.0\\bits/shared_ptr_base.h:1307
#13 0x400fa728 in ArduinoJson6172_91::Reader<fs::File, void>::read (this=0x3fff18cc) at 
<...........>
#26 ArduinoJson6172_91::deserialize<ArduinoJson6172_91::JsonDeserializer, fs::File, ArduinoJson6172_91::AllowAllFilter> (doc=..., input=..., nestingLimit=..., filter=...) at ArduinoJson/Deserialization/deserialize.hpp:68
#27 0x400faba1 in ArduinoJson6172_91::deserializeJson<fs::File> (doc=..., input=..., nestingLimit=...) at ..ArduinoJson/Json/JsonDeserializer.hpp:670
```

Issue happens when there is any file read on other task. I this case Reading Json from file. Tried to have a mutex during read & writing, but too many mutexes handling need to be call on every read and write and i cant go to that route as there is read and write file all over the place. cant find solution for this.

Anyone could shed some light on this issue? What could be the guess cause and some suggestions.
Thanks,
TS

BrianPugh, 2023-03-08:
Are you sure you are using this library? Or are you using one of the following:

1. https://github.com/lorol/LITTLEFS

2. https://github.com/espressif/arduino-esp32/tree/master/libraries/LittleFS

In the past year there have been some mutex-related bug fixes, such as #94, that may not be reflected in other projects that were based on this repo.

tsctrl, 2023-03-09:
hi @BrianPugh, yes i am using master copy of esp_littlefs and updated to the latest few days back also the mentioned wrapper above.

the LITTLEFS library was also used to wrap this library. First wrapper is arduino esp FS->LITTLEFS->esp_littlefs

BrianPugh, 2023-03-09:
cool! So lets try and isolate this issue; esp_littlefs should work as intended, you shouldn't have to explicitly add any mutexs to your read/writes.

I'm not sure that this is a concurrency issue. According to your stack trace, [this assert](https://github.com/littlefs-project/littlefs/blob/40dba4a556e0d81dfbe64301a6aa4e18ceca896c/lfs.c#L3415) is failing. Note that `LFS_ASSERT` is macro'd to `assert` from `<assert.h>`.

```
LFS_ASSERT((file->flags & LFS_O_WRONLY) == LFS_O_WRONLY);
```

So naively, it seems like you're trying to write to a file that was opened as read-only. Double check that your code isn't doing this (this is the most likely scenario). Also make sure that a single file isn't opened multiple times, [as littlefs doesn't really support this](https://github.com/littlefs-project/littlefs/issues/483#issuecomment-727597870).

If you can provide more evidence that the bug is in this port, we can try and resolve the bug. Maybe there's a subtle bug in file descriptors lookup? Probably unlikely, as it would have certainly caused more problems by now.

tsctrl, 2023-03-09:
hi @BrianPugh i updated the comment above with some question as didnt expect the reply early.

yes, i can confirm i do null check every time to open a file and close once every operation is done in the same function.

the issue arise as in this scenario:
1. i have a esp log vprintf function that route the error log to a que, this operation takes buffer and does not involve with any file operation.
2. i have a thread that open a log file with append mode and never close, check the file size before every write and do a file.write and flush operation. it close some time if the logs is full to run a write operation by emptiying the logs. this thread is responsible to receive the que and write the que to the logs.
3. i have some other thread that read other file in read mode.

based on the setup above. i am getting panic on the log write thread as it try to write on the reference of the file read on other thread. 
this issue gone if i turned off the logging thread. i have lower the logging rate, check the path every time before write operation occurs but does not fix the issue.

i would need some time to port over the codes to replicate the issue. but somehow it is great if you can point some area where i could potentially check the cause for this issue.

BrianPugh, 2023-03-09:
A minimal working example would be great. In the meantime here might be some relevant stuff:

you can manually set the log level [here](https://github.com/joltwallet/esp_littlefs/blob/fb1470a59dad5949396919f34b9856fa8d29d55f/src/esp_littlefs.c#L8) to `ESP_LOG_VERBOSE`. That will cause [logging everytime the mutex](https://github.com/joltwallet/esp_littlefs/blob/fb1470a59dad5949396919f34b9856fa8d29d55f/src/esp_littlefs.c#L714) (sometimes incorrectly referred to as a sem/semaphore in the code) is attempted to be taken/given, and then also right after it successfully occurs. The task name will also be printed. Note that we use a recursive semaphore, so the same task may take/give it several times. This aided in previous diagnoses of race conditions.

We also have [unit tests](https://github.com/joltwallet/esp_littlefs/blob/master/test/test_littlefs.c) (see README how to use; it doesn't use arduino though), some of which attempt to test for race conditions.

tsctrl, 2023-03-11:
Hi @BrianPugh,

i seem i have figured out the cause on my codes and came out with workaround for this limitation,

2 or more thread fast loop with write and read operation that can cause the issue with code as below:
```
// Thread A
File fileB;
while(;;) { 
      if(fs.exist(path)){ 
          File fileA = open(path, read); 
          if(!fileB){
           fileB = open(path, write); 
          }
          fileA.close();
          fileB.close();
      } 
}
```
```
// Thread B run a xQueueReceive with file operation below
static File fileC;
uint8_t* buff;
while(;;) {
      xQueueReceive(queue, &buff, portMAX_DELAY) == pdTRUE){
      if(fs.exist(path)){
          if(!fileC){
           fileC = open(path, write); 
          }
        fileC.write(buff);
      }
      }
}
```

as above, i notice a very fast loop in 2 or more thread where files are write and read and exist check was run simultaneously and have another files open and close in a middle of another file open cause the issue on my end. the solution i have is to reduce the loop rate with delay, avoid file exist check by caching the previous check result and try to close the file as soon as possible before another file open.
is not the real solution, as i did not know what could be the cause (i didnt try hard enough due to time) but i manage to reduce the rate of this issue from happening on my end. thank you and closing this issue.

thanks!

closed 2023-03-11 by tsctrl