joltwallet/esp_littlefs #125: Unknown CMake command "esptool_py_flash_project_args" when using self-maintained ESP-IDF

huming2207, 2023-05-23:
Hi, 

We have a fork of ESP-IDF in order to fix up some issues we have on the networking stack, and we are also using this LittleFS wrapper. The ESP-IDF fork is synced with Espressif's `master`  branch. I think there's an issue in the `project_include.cmake` script when it determines the ESP-IDF version. Here's the log when running `idf.py menuconfig`:

```
Executing action: menuconfig
Running cmake in directory /home/jackson/Projects/secret_project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/jackson/.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/jackson/Projects/secret_project"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/home/jackson/Projects/secret_project/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Component directory /home/jackson/Projects/secret_project/components/esp-protocols does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/jackson/Projects/secret_project/components/esp-protocols does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jackson/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/jackson/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/jackson/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements
......Updating lock file at /home/jackson/Projects/secret_project/dependencies.lock
Processing 1 dependencies:
[1/1] idf (5.2.0)
-- Project sdkconfig file /home/jackson/Projects/secret_project/sdkconfig
-- Compiler supported targets: xtensa-esp32s3-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/jackson/.espressif/python_env/idf5.2_py3.10_env/bin/python (found version "3.10.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "trolleyfinder-bs-main-uc" version: b15ac02-dirty
-- Adding linker script /home/jackson/Projects/secret_project/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /home/jackson/esp/esp-idf-jmh/components/soc/esp32s3/ld/esp32s3.peripherals.ld
CMake Warning at components/esp_littlefs/project_include.cmake:52 (message):
  Unsupported/unmaintained/deprecated ESP-IDF version efe919275e
Call Stack (most recent call first):
  main/CMakeLists.txt:25 (littlefs_create_partition_image)

CMake Error at components/esp_littlefs/project_include.cmake:69 (esptool_py_flash_project_args):
  Unknown CMake command "esptool_py_flash_project_args".
Call Stack (most recent call first):
  main/CMakeLists.txt:25 (littlefs_create_partition_image)

-- Configuring incomplete, errors occurred!
See also "/home/jackson/Projects/secret_project/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

```

I've added two `message()` calls in the `project_include.cmake` to log the `IDF_VER` and here's the output:

```
IDF_VER = efe919275e
IDF_VER_NO_V = efe919275e
```

Apparently it becomes the Git hash, not the actual version number. According to the documentation (see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html), it seems like the `IDF_VER` can be "set from either a version file or the Git revision of the IDF_PATH repository". Therefore we probably should use the `IDF_VERSION_MAJOR` and `IDF_VERSION_MINOR` instead. 

I will file a pull request later today to address this issue. 

Regards,
Jackson

huming2207, 2023-05-23:
Done, please review the PR, thanks!

huming2207, 2023-05-23:
We've also tried printing out the `IDF_VERSION_MAJOR` and `IDF_VERSION_MINOR` like this: 

```
message("IDF_VER_NO_V = ${IDF_VER_NO_V} ${IDF_VERSION_MAJOR} ${IDF_VERSION_MINOR}")
```

...and it outputs:

```
IDF_VER_NO_V = efe919275e 5 2
```

So I think I've done a proper fix. Please point it out if I made any mistake. Thanks.

BrianPugh, 2023-05-23:
thank you so much for identifying the bug and opening a PR! Lets continue the discussion over in the PR; I'll post a summary here once it's merged.

closed 2023-05-23 by BrianPugh

BrianPugh, 2023-05-24:
fixed in v1.5.5