joltwallet/esp_littlefs #132: Faliure to build process. "mklittlefs"

RodrigoStefanini, 2023-06-17:
Hi,
Hello, I'm trying to compile the example mentioned using the 2 import models of the component and I'm not succeeding.

I'm using ESP-IDF v5.0.2, and getting this error when building:

* FAILED: C:/Users/.../managed_components/joltwallet__littlefs/mklittlefs/mklittlefs
cmd.exe /C "cd /D C:\Users\...\managed_components\joltwallet__littlefs\mklittlefs && make clean && make dist LFS_NAME_MAX=64"
/bin/sh: -c: line 0: syntax error near unexpected token `"delims="'
/bin/sh: -c: line 0: `for /f "delims=" %%i in ('dir /s /b *.c *.cpp ^| findstr /v /i littlefs') do @echo %%i'
make: del: Command not found
make: *** [clean] Error 127
[56/895] Building C object esp-idf/esp_system.../__idf_esp_system.dir/task_wdt/task_wdt.c.objninja: build stopped: subcommand failed. 

No matter what I do, sometimes, when I don't initialize my project test example as git repository, this give an error: 

* FAILED: C:/Users/.../managed_components/joltwallet__littlefs/mklittlefs/mklittlefs
cmd.exe /C "cd /D C:\Users\...\managed_components\joltwallet__littlefs\mklittlefs && make clean && make dist LFS_NAME_MAX=64"
/bin/sh: -c: line 0: syntax error near unexpected token `"delims="'
/bin/sh: -c: line 0: `for /f "delims=" %%i in ('dir /s /b *.c *.cpp ^| findstr /v /i littlefs') do @echo %%i'
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
make: del: Command not found
make: *** [clean] Error 127
[55/874] Building C object esp-idf/esp_system...iles/__idf_esp_system.dir/crosscore_int.c.objninja: build stopped: subcommand failed.

I also try to create a thirty method using the  Component Manager Manifest File "idf_component.yml", I don't know what is the difference between it and command: "idf.py add-dependency joltwallet/littlefs==1.5.5".
>dependencies:
>    joltwallet/littlefs: "==1.5.5"
>idf:
>    version: ">=5.0.0"

Do the same error!, 

Now the Question:

Has some incompatibility with new tools of ESP-IDF v5.0.2 and LITTLEFS? I don't find any doc about, have only the managed component folder the file: "idf_component.yml" says:

>dependencies:
> idf:
>    version: '>=4.1'
>description: LittleFS is a small fail-safe filesystem for micro-controllers.
>url: https://github.com/joltwallet/esp_littlefs
>version: 1.5.5

RodrigoStefanini, 2023-06-17:
I found the cause, but I don't know why it is causing this!
If do I remove the CMakeLists.txt line "littlefs_create_partition_image(littlefs hard_disk)" it builds without any problems, 
I do all things there but of course, without mount the folder hard_disk copy inside.
My partitions CSV have the line: "littlefs,data,spiffs,,5M," and I am using esp32 with 16MB.

BrianPugh, 2023-06-17:
`littlefs_create_partition_image` creates a littlefs partition image using the external tool `mklittlefs`. It's official repo is here:

https://github.com/earlephilhower/mklittlefs

We technically use my fork, which plays nicer with this repo and has enhanced windows compatibility. If you don't need to flash files to your microcontroller's filesystem, then you don't have to use it and none of these errors will crop up. The example code uses it as a demonstration on how to use it. Clearly, if we comment out `littlefs_create_partition_image` in the CMakeLists.txt, then the file won't be written to device.

The code compiles on windows with a little bit of massaging, see this comment for some hints (some of the changes has already been incorporated into this repo):

https://github.com/joltwallet/esp_littlefs/issues/102#issue-1353190169

I don't really ever use windows, so I'm sort of at my limit in what I can support and am knowledgable about. I'd be more than happy to merge in PRs that can help make the windows compilation processes easier.

RodrigoStefanini, 2023-06-19:
How I did, I build everything in WSL, using Ubuntu-22.04, and voylahhh, it works!
I do not have admin privileges, so impossible to use [usbipd](https://github.com/dorssel/usbipd-win.git), 
I copied the folder in my workspace Windows and open with another version of vscode without WSL, only to flash and monitor the execution. 

For those in a hurry it can help!

BrianPugh, 2023-09-20:
In the newly released v1.9.0, we moved from the `mklittlefs` tool to `littlefs-python` for image generation. `littlefs-python` has a bunch of precompiled wheels for all modern operating systems, architectures, and python versions, so we won't even need to compile any image-generation tool anymore.

closed 2023-09-20 by BrianPugh