joltwallet/esp_littlefs #133: Windows support of mklittlefs

eeyrw, 2023-06-17:
When trying to use mklittlefs under Windows you may get:
```
[1/11] Generating ../../../managed_components/joltwallet__littlefs/mklittlefs/mklittlefs
FAILED: ../managed_components/joltwallet__littlefs/mklittlefs/mklittlefs C:/Users/yuan/Desktop/advanced-auto-watering/managed_components/joltwallet__littlefs/mklittlefs/mklittlefs
cmd.exe /C "cd /D C:\Users\yuan\Desktop\advanced-auto-watering\managed_components\joltwallet__littlefs\mklittlefs && make clean 
&& make dist LFS_NAME_MAX=64"
'make' 不是内部或外部命令，也不是可运行的程序
或批处理文件。
```
Is it possible to provide an executable when using Windows rather than building mklittlefs on the fly?

RodrigoStefanini, 2023-06-17:
My issues report is about that... [link](https://github.com/joltwallet/esp_littlefs/issues/132)

eeyrw, 2023-06-17:
After looking into your issue,it seems that we are talking about same things. If Windows supporting is limited or requiring other preparation, it could be better making those limitations mentioned in README.

RodrigoStefanini, 2023-06-17:
Have you tried compiling on another operating system and succeeded? Which was used?

BrianPugh, 2023-06-17:
closing this issue in favor of #132, which has more detail.

closed 2023-06-17 by BrianPugh