joltwallet/esp_littlefs #136: Missing execinfo.h

ProjectRobal, 2023-07-31:
I am trying to compile esp_littlefs under PlatformIO. I have put it in components folder in root directory of my project. I have run git command with recursive option. But it doesn't want to compile at first it complained about missing some includes from littlefs subfolders after some tweaking with esp_littlefs CMakeLists.txt I have got that error message:

*** [.pio/build/esp32-s3/libf3d/esp_littlefs/littlefs/runners/bench_runner.o] Error 1
.pio/libdeps/esp32-s3/esp_littlefs/src/littlefs/runners/test_runner.c:23:10: fatal error: execinfo.h: No such file or directory

And here is modified CMakeLists.txt:

[CMakeLists.txt](https://github.com/joltwallet/esp_littlefs/files/12221643/CMakeLists.txt)

BrianPugh, 2023-07-31:
I haven't used PlatformIO in any serious capacity, so I'm not experienced in what the issue would be. I'd find it odd that you would need to tweak the CMakeLists.txt as you did (removing `src/littlefs` from includes).

Not sure why it's unable to find `execinfo.h`, it should be provided by esp-idf.

ProjectRobal, 2023-08-01:
Sorry for problems, my platformio configuration was messed up. Apparently I had two instances of esp_littlefs in my project. After removing one instance everything is building as it should be.

closed 2023-08-01 by ProjectRobal