joltwallet/esp_littlefs #139: LittleFS failed to format.

ProjectRobal, 2023-08-17:
At first esp_littlefs had problem with writting/reading to files, I don't know if it was because I was using in fread/fwrite in static functions. Then I have used erase flash in PlatformIO and since then I am getting this error no matter what I do. I am using
ESP-32-S3 with 16MB flash size and 8MB PSRAM.

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1704
load:0x403c9700,len:0xc48
load:0x403cc700,len:0x3200
entry 0x403c9974
I (24) boot: ESP-IDF 5.0.2 2nd stage bootloader
I (25) boot: compile time 00:32:47
I (25) boot: chip revision: v0.1
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 16MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (72) boot:  2 app0             OTA app          00 10 00010000 00640000
I (79) boot:  3 app1             OTA app          00 11 00650000 00640000
I (87) boot:  4 littlefs         Unknown data     01 82 00c90000 002b2400
I (94) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=3b638h (243256) map
I (148) esp_image: segment 1: paddr=0004b660 vaddr=3fc98500 size=049b8h ( 18872) load
I (152) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=c5534h (808244) map
I (292) esp_image: segment 3: paddr=0011555c vaddr=3fc9ceb8 size=0f124h ( 61732) load
I (304) esp_image: segment 4: paddr=00124688 vaddr=40374000 size=1449ch ( 83100) load
I (332) boot: Loaded app from partition at offset 0x10000
I (332) boot: Disabling RNG early entropy source...
I (344) octal_psram: vendor id    : 0x0d (AP)
I (344) octal_psram: dev id       : 0x02 (generation 3)
I (344) octal_psram: density      : 0x03 (64 Mbit)
I (349) octal_psram: good-die     : 0x01 (Pass)
I (354) octal_psram: Latency      : 0x01 (Fixed)
I (359) octal_psram: VCC          : 0x01 (3V)
I (364) octal_psram: SRF          : 0x01 (Fast Refresh)
I (370) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (376) octal_psram: BurstLen     : 0x01 (32 Byte)
I (381) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (388) octal_psram: DriveStrength: 0x00 (1/1)
I (393) esp_psram: Found 8MB PSRAM device
I (397) esp_psram: Speed: 40MHz
I (401) cpu_start: Pro cpu up.
I (405) cpu_start: Starting app cpu, entry point is 0x403763b8
I (0) cpu_start: App cpu up.
I (1142) esp_psram: SPI SRAM memory test OK
I (1152) cpu_start: Pro cpu start user code
I (1152) cpu_start: cpu freq: 160000000 Hz
I (1152) cpu_start: Application information:
I (1155) cpu_start: Project name:     GaraiCPU
I (1161) cpu_start: App version:      4ec7074-dirty
I (1166) cpu_start: Compile time:     Aug 18 2023 00:41:44
I (1172) cpu_start: ELF file SHA256:  875a18522a53975c...
I (1178) cpu_start: ESP-IDF:          5.0.2
I (1183) cpu_start: Min chip rev:     v0.0
I (1188) cpu_start: Max chip rev:     v0.99 
I (1193) cpu_start: Chip rev:         v0.1
I (1198) heap_init: Initializing. RAM available for dynamic allocation:
I (1205) heap_init: At 3FCB1200 len 00038510 (225 KiB): DRAM
I (1211) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1218) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1224) heap_init: At 600FE020 len 00001FE0 (7 KiB): RTCRAM
I (1232) spi_flash: detected chip: gd
I (1235) spi_flash: flash io: dio
W (1240) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (1250) sleep: Configure to isolate all GPIO pins in sleep state
I (1255) sleep: Enable automatic switching of GPIO sleep configuration
I (1262) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1283) MAIN: Hello World!
Heap summary for capabilities 0x00000004:
  At 0x3fcb1200 len 230672 free 205472 allocated 23512 min_free 205472
    largest_free_block 204788 alloc_blocks 46 free_blocks 3 total_blocks 49
  At 0x3fce9710 len 22308 free 21560 allocated 0 min_free 21560
    largest_free_block 21492 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3fcf0000 len 32768 free 32020 allocated 0 min_free 32020
    largest_free_block 31732 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x600fe020 len 8160 free 7768 allocated 0 min_free 7768
    largest_free_block 7668 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 266820 allocated 23512 min_free 266820 largest_free_block 204788
E (1343) esp_littlefs: components/esp_littlefs/src/littlefs/lfs.c:1333:error: Corrupted dir pair at {0x0, 0x1}

W (1353) esp_littlefs: mount failed,  (-84). formatting...

What can be responsible for it?

BrianPugh, 2023-08-18:
can you share your code?

ProjectRobal, 2023-08-18:
Here is a link to a repository: 

https://github.com/ProjectRobal/GaraiSumo

BrianPugh, 2023-08-18:
on initial skim, things look relatively sane.

I've only ever really tested this library on an esp32; but I should really expand my collection, I just ordered an esp32-s3 board; should arrive tomorrow.

ProjectRobal, 2023-08-18:
SPIFFS also fails. I just tested it.

BrianPugh, 2023-08-18:
hmmm I see that you are using platform.io. I don't have a lot of platform.io experience, but it looks like your board isn't setup properly. Checkout [this page](https://docs.platformio.org/en/latest/platforms/espressif32.html).

Naively, it looks like you should add  the following to your `platformio.ini`:

```
board_build.flash_mode = qio
build_flags =
    -DCONFIG_SPIRAM_CACHE_WORKAROUND
```

ProjectRobal, 2023-08-18:
It doesn't change much,

BrianPugh, 2023-09-20:
closing issue due to inactivity, probably some platformio misconfiguration.

closed 2023-09-20 by BrianPugh