joltwallet/esp_littlefs #153: LittleFS 1.10.1 Starving in place a FatFS wear leaveling

filzek, 2023-10-19:
Hi,

we are using the latest version of 1.10.1 with idf 4.4.6 and when we replace the FatFS with LittleFS the system dispare a lot of wdts everywhere, so, something is intense consuming cpu usage during flash operations, we have to add a lot of task wait to clear the wdts.

Is that normal or a buggy?

In the FatFS everything works perfectly without any wdt.

The LittleFS doesnt dispare any WDT reset only intense wdt usage.

BrianPugh, 2023-10-19:
There are a few time/cpu expensive operations within `littlefs`. It seems like your current issue is that some `littlefs` operations take up too much time, and consequently trigger your watch dog timers?

If so, I suppose there are a few options:
1. Lengthen your WDT period via [`CONFIG_ESP_TASK_WDT_TIMEOUT_S`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp-task-wdt-timeout-s).
2. Call `taskYIELD()` before filesystem operations.
3. Upstream littlefs has a function [`lfs_fs_gc`](https://github.com/littlefs-project/littlefs/blob/f77214d1f0a8ccd7ddc7e1204fedd25ee5a41534/lfs.h#L725), that can front-load some of the expensive operations. I could add an interface to expose this, and the user could call this during times the CPU would otherwise be idling.

filzek, 2023-11-07:
did not solve the problem, the system got corrupted everywhere in 1.10.1, we will try the 1.10.2 version

BrianPugh, 2023-11-07:
Can you explain your usage context a bit more? Also have you tried debugging to see what is causing your system to hang?

BrianPugh, 2024-02-25:
Closing due to lack of activity.

closed 2024-02-25 by BrianPugh