joltwallet/esp_littlefs #155: Failed to build "littlefs_py_venv"

BoltSwith, 2023-10-20:
Greetings,

Great library...!!   However, I haven't been able to successfully compile it yet...

After including the Component to my components folder, and editing my partition.csv file accordingly, each attempt at building the project produces the error shown below even after cleaning the project. Kindly point me to the source of the error.

**ERROR MESSAGE:**

-- Configuring done
-- Generating done
-- Build files have been written to: D:/ESP_PROJ/METERBIT_FRAME_1/build

 *  Executing task: ninja  

[3/1265] Generating ../../littlefs_py_venv
FAILED: littlefs_py_venv D:/ESP_PROJ/METERBIT_FRAME_1/build/littlefs_py_venv
cmd.exe /C "cd /D D:\ESP_PROJ\METERBIT_FRAME_1\build\esp-idf\main && D:\ESP\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe -m venv D:/ESP_PROJ/METERBIT_FRAME_1/build/littlefs_py_venv && D:/ESP_PROJ/METERBIT_FRAME_1/build/littlefs_py_venv/Scripts/pip.exe install -r D:/ESP_PROJ/METERBIT_FRAME_1/components/joltwallet__littlefs/image-building-requirements.txt"
D:\ESP\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe: No module named venv.__main__; 'venv' is a package and cannot be directly executed
[11/1265] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************

BrianPugh, 2023-10-20:
hmmmm, that's a bit odd (also I don't have a strong intuition on Windows issues), but is there an erroneous `venv` folder somewhere? Basically it's having trouble creating a python virtual environment.

ptr224, 2024-01-17:
I solved this by changing it with virtualenv, in project_include.cmake:33, after installing it via pip.
I don't know if it may be useful to anyone else so I'm sharing it here.
Is this module supposed to be used with a generic python installation?

BrianPugh, 2024-01-17:
[venv](https://docs.python.org/3/library/venv.html) is supposed to be a python builtin, but sometimes it's not shipped with python (for example, on ubuntu/debian it has to be additionally installed with `sudo apt install python3-venv`). The builtin `venv` is basically a subset of the thirdparty pacakge `virtualenv`.

BrianPugh, 2024-01-17:
as for @BoltSwith issue, I'm pretty sure they have an erroneous `venv` folder in their working directory with an `__init__.py` file inside of it. Reproducing the issue on macos:

```
# demonstrating that the venv module is correctly parsed
$ python -m venv
usage: venv [-h] [--system-site-packages] [--symlinks | --copies] [--clear] [--upgrade] [--without-pip]
            [--prompt PROMPT] [--upgrade-deps]
            ENV_DIR [ENV_DIR ...]
venv: error: the following arguments are required: ENV_DIR

# creating a bad venv folder in the current working directory
$ mkdir venv && touch venv/__init__.py

# Rerunning the command
$ $ python -m venv
/Users/brianpugh/projects/esp_littlefs/.venv/bin/python: No module named venv.__main__; 'venv' is a package and cannot be directly executed
```

ptr224, 2024-01-17:
I confirm that the builtin idf python doesn't have it though
I guess it's a vscode extension problem

`D:\ptr22\Source\EspIdf\test>python -m venv
C:\Users\ptr22\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe: No module named venv.__main__; 'venv' is a package and cannot be directly executed
`

BrianPugh, 2024-01-18:
I don't have a good windows environment setup right now, can you check if there's a stray `venv` folder... somewhere? I can't tell what the working directory is based on this context.

ptr224, 2024-01-18:
I confirm that there is no venv directory neither in the project nor in the environment
Well actually there is one, but it's not in the used environment.
In fact I checked the IDF windows installer code, it looks like they do ship a custom python, while they don't for macOS and Linux.
This custom python then creates a virtual environment from where everything actually runs and there is no venv script there.
I guess it is a Windows only problem then.
Probably a full python setup instead of the embedded one would solve the problem?
Either this or using virtualenv.