joltwallet/esp_littlefs #157: Failed to build, error with `block_count`

guo-max, 2023-10-26:
Got this error when try to build with version 1.10.2 from master branch. 
[image]

``` 
   void get_total_and_used_bytes(esp_littlefs_t *efs, size_t *total_bytes, size_t *used_bytes) {
    sem_take(efs);
    size_t total_bytes_local = efs->cfg.block_size * efs->fs->block_count;
  ...
```

I checked the early version, before Version 1.10.0, this function is like this
```
void get_total_and_used_bytes(esp_littlefs_t *efs, size_t *total_bytes, size_t *used_bytes) {
    sem_take(efs);
    size_t total_bytes_local = efs->cfg.block_size * efs->cfg.block_count;
   ...
 }
```
Is this a bug?

BrianPugh, 2023-10-26:
are you building this from cloning this repo? Make sure to run a `git submodule update`

guo-max, 2023-10-26:
> are you building this from cloning this repo? Make sure to run a `git submodule update`

Thanks for your quick reply. I downloaded the `zip` file. I will try to clone and update the submodule.

guo-max, 2023-10-26:
issue fixed after update submodule.
Thanks again.

closed 2023-10-26 by guo-max