joltwallet/esp_littlefs #163: How to use directories? - Correctly include libary

j54j6, 2023-12-14:
Hey, thank you for your great work and the port to ESP IDF. I am very new to ESP IDF, due to the lack of support of the esp32-C6 I want to  use esp idf with littleFS from now on instead of arduino.

I have managed to get the example working no problems there, but if I want to use the "normal" functions like creating a directory or opening a directory it(Ninja/VSCode) cannot find these functions during compilation. 

For reference I have the following code (this code is shorted) 

```
#include "esp_err.h"
#include "esp_log.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"

#include <string.h>
#include <stdio.h>
#include <sys/stat.h>
#include <unistd.h>
#include "esp_idf_version.h"
#include "esp_flash.h"

#if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0)
#include "esp_chip_info.h"
#include "spi_flash_mmap.h"
#endif

#include "esp_littlefs.h"

static const char *TAG = "demo_esp_littlefs";

void getSerialFileStructure(const char* path)
{
    recursive(path);
}

void recursive(const char* path)
{
 esp_vfs_littlefs_conf_t conf = {
            .base_path = "/littlefs",
            .partition_label = "littlefs",
            .format_if_mount_failed = true,
            .dont_mount = false,
        };

  static int insideDirs = 0;
  static const char* mainPath = "";
  const char* oldPath = mainPath;
  mainPath = path;
  vfs_littlefs_dir_t dir = vfs_littlefs_opendir(conf, path);
  //int number = getNumberOfFiles(path);
  int actual = 0;
  return;
  return; //Shorted 
}

 void app_main()
{
    <<The example Code out of this repo>>

     //This function is used to show all the data on the filesystem - Exported from an old Arduino FS Manager Libary
    getSerialFileStructure("/"); 

// All done, unmount partition and disable LittleFS
        esp_vfs_littlefs_unregister(conf.partition_label);
        ESP_LOGI(TAG, "LittleFS unmounted");
}
```

If I use this Code it can not be compiled - I get this Error:
`
error: unknown type name 'vfs_littlefs_dir_t'; did you mean 'esp_vfs_littlefs_conf_t'?
   47 |   vfs_littlefs_dir_t dir = vfs_littlefs_opendir(path);
      |   ^~~~~~~~~~~~~~~~~~
      |   esp_vfs_littlefs_conf_t
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/main/ESP-C6_OS.c:47:28: error: implicit declaration of function 'vfs_littlefs_opendir' [-Werror=implicit-function-declaration]
   47 |   vfs_littlefs_dir_t dir = vfs_littlefs_opendir(path); `

But if I use VSCode and F12 (go to definition) on the type (`vfs_littlefs_dir_t` ) it goes into esp_littlefs.c and shows it, but if I want to seek the definition of the function (`vfs_littlefs_opendir(void* ctx, const char* name)`) it cannot find it (the function is also in esp_littlefs.c), so the files are all there and I guess IDF managed components included it right... 

I am not sure what I do wrong and I guess this is not a general issue with the libary because the example is working but If I use the offical spiffs configuration (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spiffs.html) - I cannot find anything (and as far as I know directories are not supported in spiffs - all files are created with "/" in their name, if provided) - I have no clue where to start, there is no issue like mine here (or I didn't find it so I guess it is a problem on my side an I miss something very particular ^^

I would be very thankful if anyone can help me get started and show me, why the functions can not be used. SDK Config says Direcotires are supported (value is 1). Only thing I found all these functions are only in the .c file but not in the corresponding header file.

Thank you for your time and help :)
j54j6

BrianPugh, 2023-12-14:
Hey @j54j6 ! You're not supposed to use any of the `vfs_littlefs_*` functions directly. What you do is (this also goes for other filesystems like spiffs or FAT):

1. Define your `esp_vfs_littlefs_conf_t conf;`.
2. Register your configured filesystem with `esp_vfs_spiffs_register(conf);`
3. Now that the filesystem has been registered, you can interact with the filesystem using normal C File operations like `open` and `opendir`.

Let me know if you have further questions!

j54j6, 2023-12-15:
Hi @BrianPugh, 

thank you so much for your help. I've got it working <3
If there are any further questions I will open a new issue :) - Thank you for your time, hope you have a wonderful day^^

closed 2023-12-15 by j54j6