joltwallet/esp_littlefs #164: Accept pull request to add library.json in root ?

EvEggelen, 2023-12-17:
To support PlatformIO ini files with IDF framework, adding a library.json would help.
With the following line you could add the library to the project : lib_deps = https://github.com/joltwallet/esp_littlefs.git

To my knowledge this can be done by adding a library.json in the root of the esp_littlefs repository. The content would like something like this:

```
{
  "name": "esp_littlefs",
  "version": "0.0.0+20231217184442",
  "build": {
    "srcFilter": "+<*> -<littlefs/runners> -<littlefs/benches> -<littlefs/tests>",
    "flags": [
      "-I ./src/littlefs/"
    ]
  }
}
```

When successfully tested, would you accept a pull request ?

BrianPugh, 2023-12-17:
I don't use PlatformIO, but sounds like it'd be useful to many people! I'd merge that PR.

Just make sure to also update [.bumpversion.cfg](https://github.com/joltwallet/esp_littlefs/blob/master/.bumpversion.cfg)!

EvEggelen, 2023-12-18:
O.K. I will clone the repository and give it a try.

EvEggelen, 2023-12-18:
I cloned the repository and added the library.json file. You can now use it within PlatformIO by added the following line the to platform.ini file. `lib_deps = https://github.com/EvEggelen/esp_littlefs.git`

With this you still need to move / manually move the Kconfig file to the src directory. 
When done you can run the command inside platformIO : `~/.platformio/penv/bin/pio run -t menuconfig`

However I am trying to figure out how to get this component registered correctly and work out of the box with menuconfig. Do you have experience how to do this ?

I tried the following:
file(GLOB SOURCES src/littlefs/*.c)
list(APPEND SOURCES src/esp_littlefs.c src/littlefs_api.c src/lfs_config.c)
**list(APPEND SOURCES Kconfig)**

if(IDF_VERSION_MAJOR GREATER_EQUAL 5)
    list(APPEND pub_requires esp_partition)
else()
    list(APPEND pub_requires spi_flash)
endif()
list(APPEND priv_requires esptool_py spi_flash vfs)

idf_component_register(
    SRCS ${SOURCES}
    INCLUDE_DIRS include
    PRIV_INCLUDE_DIRS src
    REQUIRES ${pub_requires}
    PRIV_REQUIRES ${priv_requires}
)

BrianPugh, 2023-12-18:
So I don't have any real experience with PlatformIO, so I cannot add much input/assistantce.

EvEggelen, 2023-12-18:
Would it be a problem for your project to rename the Kconfig to Kconfig.projbuild ? Or can you live with both Kconfig and Kconfig.projbuild ?

BrianPugh, 2023-12-18:
so long as it still works with esp-idf, I'm fine with Kconfig.projbuild.

EvEggelen, 2023-12-18:
I just fixed the issue. Now only the original Kconfig is sufficient. So please review my additions in the the README.md file.
I tested the changes and everything seems to work O.K. with the latest PlatformIO and your lib. 

The only 'issue' is a static version number in the library.json. Currently I set it to : "version": "0.0.0". Do you know a trick to automatically set this to the official version number  ( so without editing the file every time ) ?

BrianPugh, 2023-12-18:
Can you open up a PR for me to review?

For versioning, this is what I was[ referring to here](https://github.com/joltwallet/esp_littlefs/issues/164#issuecomment-1859296628) about the bumpversion config. Basically just reference it in that file, and set the current version in the file so that bumpversion will automatically update it.

EvEggelen, 2023-12-19:
Never use the bumpversion tool ( used others) before. But indeed this is what I was looking for. Hope that I made the changes correctly. 

I created a pull request. Please review before merging. Let me know any remarks or comments.

BrianPugh, 2023-12-20:
merged! this is a part of v1.12.0

closed 2023-12-20 by BrianPugh

EvEggelen, 2023-12-20:
Great. However I noticed that you changed the command for running the menuconfig. If you use a standard install ( install Visual Studio Code, within Visual Studio code install PlatformIO extension. En then within PlatformIO install IDF framework) , pio is NOT put in the search path ( and triggers an advice to install it). IDF within PlatformIO has it quirks. So I recommend that you mention in the README.md that with a default install you need to use the following command in the terminal within PlatformIO: `~/.platformio/penv/bin/pio run -t menuconfig`.

BrianPugh, 2023-12-20:
Aren't you supposed to add `~/.platformio/penv/bin/` to your `PATH`?

EvEggelen, 2023-12-20:
I normally expect that the installer takes care of this so you do not need to manually add it ( as the rest is automated).  As I wanted to lower the threshold for using this lib, I added it. If you do not know better, and follow the message to install pio with apt-get, you run the risk ending up version conflicts.

I do not add it to the PATH as the directory is managed by PlatformIO platform installer. The developers willing to add this to the PATH also know to remove the directory from the command. When uninstalling the platform from PlatformIO you do not need to  forget to manually remove it. 

Anyway, this is what I though was best to do. It is not black and white.

BrianPugh, 2023-12-20:
Looking into it, this is PlatformIO's virtual environment, so you're really supposed to activate it (`source ~/.platformio/penv/bin/activate`), which appropriately add its to the path. I think the [CLI/terminal button](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) does this for you (but I've never used it, so I don't personally know). I don't believe it's appropriate to directly execute `~/.platformio/penv/bin/pio`.

EvEggelen, 2023-12-20:
You are right. If you use the terminal button it is indeed activated. However if you do terminal -> new it is not. Also the default terminal that is created is also not activated. So I learned something today. Thanks ! Maybe we should mention this as a note.