joltwallet/esp_littlefs #171: [question] File creation while iterating

danielbuechele, 2024-01-18:
I'm iterating a directory using `readdir`. What is the behavior when another task creates a new file while I'm iterating in the first task. Will the new file be "invisible", will I iterate the newly created file or is the behavior unspecified?

Or is this a question I should rather directly ask on the littleFS repo?

BrianPugh, 2024-01-18:
I *think* this is an upstream question. Basically the question is: "if a file is created after lfs_dir_open is called, is it included in the traversal?"

danielbuechele, 2024-01-18:
Yeah, I thought so. I did some tests myself and it looks like files are generally iterated in alphabetical order. If the newly created file has a filename that is later in the alphabet it would be included in the iteration, if my loop was already past the point it wouldn't be included.

Thanks for your quick reply!

closed 2024-01-18 by danielbuechele