joltwallet/esp_littlefs #173: is it possible to support larger sdcard by changing SIZE_MAX

wang33winner, 2024-01-26:
if i use a sdcard larger than 4GiB, ther will be a warinning says "SD card is too big ..."
and i checked the code is here:
```c
        size_t max_allowed_blk_cnt = 0;
        if (((uint64_t)sdcard->csd.capacity * (uint64_t)sdcard->csd.sector_size) > SIZE_MAX) {
            max_allowed_blk_cnt = SIZE_MAX / sdcard->csd.sector_size;
            ESP_LOGW(ESP_LITTLEFS_TAG, "SD card is too big (sector=%d, count=%d; total=%llu bytes), throttling to maximum possible %u blocks",
                     sdcard->csd.sector_size, sdcard->csd.capacity,
                     (((uint64_t)sdcard->csd.capacity * (uint64_t)sdcard->csd.sector_size)), max_allowed_blk_cnt);
        } else {
            max_allowed_blk_cnt = sdcard->csd.capacity;
        }
```
so is it possible to change SIZE_MAX to 0xffffffffffffffffull?
maybe some changes should also to be done in `get_total_and_used_bytes`，is there any other limits for SIZE_MAX ?

BrianPugh, 2024-01-26:
LittleFS previously claimed to somewhat support up-to 4GB filesystems, but this has been [formally rolled back in the latest release v2.9](https://github.com/littlefs-project/littlefs/releases/tag/v2.9.0) (which I should update & released a new version of esp_littlefs) to 2GB. [The >2GB functionality wasn't well tested.](https://github.com/littlefs-project/littlefs/pull/886#discussion_r1376604881) So, if I were you, I wouldn't attempt to use a partition larger than 2GB with littlefs.

huming2207, 2024-01-26:
Yea @BrianPugh is correct. 

Plus this port also uses 32-bit counters & calculations everywhere (e.g. on block counts etc.), I don't think it will work even if LittleFS itself support something bigger than 2GB (or 4GB). 

However I think it's worthy to discuss about using or even creating partitions (particularly GPT partitions, or at very least MBR) on SD card. Otherwise there's no way to make the full use on any modern SD card bigger than 2GB.

huming2207, 2024-01-29:
I've just had a look on this and I think porting MBR is definitely possible. On top of that GPT partition table is also doable but it's a bit overkill for microcontrollers. 😅

Meanwhile [esp_partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/partition.html) is also usable, but needs quite a bit hacks. The size and address fields in `esp_partition` are all `uint32_t` which is limited to 4GB. I guess Espressif initially only consider using this library within their SPI flash scope, not for something bigger like SD cards.

BrianPugh, 2024-01-31:
Hey guys, going to close this issue here as there's no real action items for this repo. You may continue the conversation if you'd like, however 😄

closed 2024-01-31 by BrianPugh

adokitkat, 2024-02-06:
> Meanwhile [esp_partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/partition.html) is also usable, but needs quite a bit hacks. The size and address fields [...]

Hello :) If you want to support large storage media or make something easier for you from our side, discussion and pull requests are welcome!

huming2207, 2024-02-06:
Hi @adokitkat thanks for your comment! 

I will write up a feature request for now at ESP-IDF repo for further discussion about this. Meanwhile I'm busy working on something else so I'm afraid I don't have time at this stage for working on a PR.