joltwallet/esp_littlefs #178: Can not delete open file

voedipus, 2024-02-22:
What is the reason for this?
https://github.com/joltwallet/esp_littlefs/blob/41873c20fb5cdbcf28d7d6cc04e4bcb4a1305317/src/esp_littlefs.c#L1882

BrianPugh, 2024-02-22:
I'm  not sure the exact inner workings of littlefs, but I imagine writing/reading from a deleted file may lead to corrupt results (either in memory, or on disk).

voedipus, 2024-03-12:
> I'm not sure the exact inner workings of littlefs, but I imagine writing/reading from a deleted file may lead to corrupt results (either in memory, or on disk).

No. I open a file on STM32 and without closing the file delete it. No data is corrupted. But in this wrapper it is not possible to delete file without closing it first.

BrianPugh, 2024-03-12:
So from your experience, and [this comment by geky,](https://github.com/littlefs-project/littlefs/issues/842) it seems like it might be ok?

Would you be available to open up a PR along with a unit test for this?

I worry that some logic in `vfs_littlefs_close` will also need to be changed. For example, if a file is deleted before it's closed, the file handle can never be forfeited with the current logic. Maybe the filehandle also needs to be [freed](https://github.com/joltwallet/esp_littlefs/blob/41873c20fb5cdbcf28d7d6cc04e4bcb4a1305317/src/esp_littlefs.c#L1717) in the error-handling logic.